apache/nuttx #11077: Documentation: port remaining README.txt in platforms

raiden00pl, 2023-10-28:
README.txt from boards have been moved to Documentation and (almost all) included in rst files. Some were migrated to rst, but many remained in txt format.

This problem is mentioned in https://github.com/apache/nuttx/issues/9094 as a task, but there's a lot of work to be done, so it's better to keep it as a separate issue.

```
[raiden00:...ttx/Documentation/platforms]$ find -name "README.*"                                                                                                                                                                                                      (master✱) 
./misco/lm32/boards/misoc/README.txt
./xtensa/esp32/boards/esp32-ethernet-kit/README.txt
./arm/lpc31xx/boards/ea3152/README.txt
./arm/lpc31xx/boards/olimex-lpc-h3131/README.txt
./arm/lpc31xx/boards/ea3131/README.txt
./arm/sam34/boards/arduino-due/README.txt
./arm/sam34/boards/flipnclick-sam3x/README.txt
./arm/sam34/boards/sam4s-xplained-pro/README.txt
./arm/sam34/boards/sam4s-xplained/README.txt
./arm/sam34/boards/sam4l-xplained/README.txt
./arm/sam34/boards/sam3u-ek/README.txt
./arm/sam34/boards/sam4e-ek/README.txt
./arm/sam34/boards/sam4cmp-db/README.txt
./arm/efm32/boards/olimex-efm32g880f128-stk/README.txt
./arm/efm32/boards/efm32gg-stk3700/README.txt
./arm/efm32/boards/efm32-g8xx-stk/README.txt
./arm/lpc40xx/boards/lpc4088-quickstart/README.txt
./arm/lpc40xx/boards/lpc4088-devkit/README.txt
./arm/lpc40xx/boards/lx_cpu/README.txt
./arm/tms570/boards/launchxl-tms57004/README.txt
./arm/tms570/boards/tms570ls31x-usb-kit/README.txt
./arm/kl/boards/teensy-lc/README.txt
./arm/kl/boards/freedom-kl26z/README.txt
./arm/kl/boards/freedom-kl25z/README.txt
./arm/max326xx/boards/max32660-evsys/README.txt
./arm/nuc1xx/boards/nutiny-nuc120/README.txt
./arm/at32/boards/at32f437-mini/README.txt
./arm/samv7/boards/samv71-xult/README.txt
./arm/samv7/boards/same70-qmtech/README.txt
./arm/samv7/boards/same70-xplained/README.txt
./arm/str71x/boards/olimex-strp711/README.txt
./arm/kinetis/boards/teensy-3.x/README.txt
./arm/kinetis/boards/freedom-k28f/README.txt
./arm/kinetis/boards/freedom-k66f/README.txt
./arm/kinetis/boards/kwikstik-k40/README.txt
./arm/kinetis/boards/twr-k60n512/README.txt
./arm/kinetis/boards/freedom-k64f/README.txt
./arm/kinetis/boards/twr-k64f120m/README.txt
./arm/tiva/boards/lm3s8962-ek/README.txt
./arm/tiva/boards/lm4f120-launchpad/README.txt
./arm/tiva/boards/ekk-lm3s9b96/README.txt
./arm/tiva/boards/tm4c123g-launchpad/README.txt
./arm/tiva/boards/launchxl-cc1312r1/README.txt
./arm/tiva/boards/eagle100/README.txt
./arm/tiva/boards/lm3s6432-s2e/README.txt
./arm/tiva/boards/tm4c1294-launchpad/README.txt
./arm/tiva/boards/tm4c129e-launchpad/README.txt
./arm/tiva/boards/launchxl-cc1310/README.txt
./arm/tiva/boards/lm3s6965-ek/README.txt
./arm/tiva/boards/dk-tm4c129x/README.txt
./arm/lpc54xx/boards/lpcxpresso-lpc54628/README.txt
./arm/lpc214x/boards/mcu123-lpc214x/README.txt
./arm/lpc214x/boards/zp214xpa/README.txt
./arm/qemu/boards/qemu-armv7a/README.txt
./arm/eoss3/boards/quickfeather/README.txt
./arm/lpc2378/boards/olimex-lpc2378/README.txt
./arm/fvp-v8r-aarch32/boards/fvp-armv8r-aarch32/README.txt
./arm/am335x/boards/beaglebone-black/README.txt
./arm/dm320/boards/ntosd-dm320/README.txt
./arm/samd2l2/boards/samd20-xplained/README.txt
./arm/samd2l2/boards/saml21-xplained/README.txt
./arm/samd2l2/boards/arduino-m0/README.txt
./arm/samd2l2/boards/samd21-xplained/README.txt
./arm/mx8mp/boards/verdin-mx8mp/README.txt
./arm/a1x/boards/pcduino-a10/README.txt
./arm/lc823450/boards/lc823450-xgevk/README.txt
./arm/sama5/boards/sama5d2-xult/README.txt
./arm/sama5/boards/sama5d3-xplained/README.txt
./arm/sama5/boards/sama5d3x-ek/README.txt
./arm/sama5/boards/sama5d4-ek/README.txt
./arm/imx6/boards/sabre-6quad/README.txt
./arm/cxd56xx/boards/spresense/README.txt
./arm/lpc43xx/boards/lpc4357-evb/README.txt
./arm/lpc43xx/boards/lpc4370-link2/README.txt
./arm/lpc43xx/boards/lpc4337-ws/README.txt
./arm/lpc43xx/boards/lpc4330-xplorer/README.txt
./arm/lpc43xx/boards/bambino-200e/README.txt
./arm/lpc17xx/boards/mcb1700/README.txt
./arm/lpc17xx/boards/lpcxpresso-lpc1768/README.txt
./arm/lpc17xx/boards/zkit-arm-1769/README.txt
./arm/lpc17xx/boards/mbed/README.txt
./arm/lpc17xx/boards/lincoln60/README.txt
./arm/lpc17xx/boards/olimex-lpc1766stk/README.txt
./arm/lpc17xx/boards/open1788/README.txt
./arm/lpc17xx/boards/u-blox-c027/README.txt
./arm/lpc17xx/boards/pnev5180b/README.txt
./arm/rp2040/boards/waveshare-rp2040-lcd-1.28/README.txt
./arm/rp2040/boards/adafruit-feather-rp2040/README.txt
./arm/rp2040/boards/raspberrypi-pico/README.txt
./arm/rp2040/boards/pimoroni-tiny2040/README.txt
./arm/rp2040/boards/adafruit-kb2040/README.txt
./arm/rp2040/boards/adafruit-qt-py-rp2040/README.txt
./arm/rp2040/boards/raspberrypi-pico-w/README.txt
./arm/c5471/boards/c5471evm/README.txt
./arm/samd5e5/boards/metro-m4/README.txt
./arm/samd5e5/boards/same54-xplained-pro/README.txt
./mips/pic32mz/boards/flipnclick-pic32mz/README.txt
./mips/pic32mz/boards/pic32mz-starterkit/README.txt
./mips/pic32mz/boards/chipkit-wifire/README.txt
./mips/pic32mx/boards/pic32mx-starterkit/README.txt
./mips/pic32mx/boards/mirtoo/README.txt
./mips/pic32mx/boards/sure-pic32mx/README.txt
./mips/pic32mx/boards/pic32mx7mmb/README.txt
./mips/pic32mx/boards/ubw32/README.txt
./x86/qemu/boards/qemu-i486/README.txt
./or1k/mor1kx/boards/or1k/README.txt
./arm64/fvp-v8r/boards/fvp-armv8r/README.txt
./arm64/qemu/boards/qemu-armv8a/README.txt
./arm64/imx8/boards/imx8qm-mek/README.txt
./risc-v/qemu-rv/boards/rv-virt/README.txt
./risc-v/k210/boards/maix-bit/README.txt
./risc-v/fe310/boards/hifive1-revb/README.txt
./risc-v/litex/boards/arty_a7/README.txt
./risc-v/esp32c3/boards/esp32c3-devkit-rust-1/README.txt
./risc-v/c906/boards/smartl-c906/README.txt
./risc-v/hpm6750/boards/hpm6750evk2/README.txt
./risc-v/rv32m1/boards/rv32m1-vega/README.txt
./hc/m9s12/boards/demo9s12ne64/README.txt
./hc/m9s12/boards/ne64badge/README.txt
./x86_64/qemu/boards/qemu-intel64/README.txt
./avr/at90usb/boards/micropendous3/README.txt
./avr/at90usb/boards/teensy-2.0/README.txt
./avr/at32uc3/boards/avr32dev1/README.txt
./avr/atmega/boards/arduino-mega2560/README.txt
./avr/atmega/boards/amber/README.txt
./avr/atmega/boards/moteino-mega/README.txt
./z16/z16f/boards/z16f2800100zcog/README.txt
./sim/sim/boards/sim/README.txt
./renesas/sh1/boards/us7032evb1/README.txt
./renesas/m16c/boards/skp16c26/README.txt
./renesas/rx65n/boards/rx65n-grrose/README.txt
./renesas/rx65n/boards/rx65n-rsk2mb/README.txt
./z80/ez80/boards/ez80f910200zco/README.txt
./z80/ez80/boards/ez80f910200kitg/README.txt
./z80/ez80/boards/z20x/README.txt
./z80/ez80/boards/makerlisp/README.txt
./z80/z8/boards/z8f64200100kit/README.txt
./z80/z8/boards/z8encore000zco/README.txt
./z80/z80/boards/z80sim/README.txt
./z80/z180/boards/p112/README.txt

```

linguini1, 2025-05-07:
It looks like all the RP2040 boards can be checked off from this list.

linguini1, 2025-05-25:
Complete in #16439 
- avr/at90usb/boards/micropendous3/README.txt
- avr/at90usb/boards/teensy-2.0/README.txt

linguini1, 2025-05-27:
Completed in #16449:

* arm64/fvp-v8r/boards/fvp-armv8r/README.txt
* arm64/qemu/boards/qemu-armv8a/README.txt
* arm64/imx8/boards/imx8qm-mek/README.txt

raiden00pl, 2025-09-24:
I cleared the list of already ported pages and I changed the order according to `$ find -name "README.txt"`