apache/nuttx #11079: Documentation: better look for Supported Platforms

raiden00pl, 2023-10-28:
At the moment it's just a long list of supported architectures, supported chip families and supported boards.

```
* architecture
** chip familly
*** board 1
*** board 2
*** ....
```

Maybe someone with front-end skills has an idea on how to make it more readable and pleasing to the eye (with Sphinx) ?

acassis, 2023-10-28:
@raiden00pl I think we could create a skeleton/template that people could use when submitted new board/architecture documentation. Many boards are missing pictures, supported peripherals, instructions to flash the board, etc

raiden00pl, 2023-10-29:
@acassis this is another problem for which I created a separate issue https://github.com/apache/nuttx/issues/11080

In this issue, I'm talking more about this page https://nuttx.apache.org/docs/latest/platforms/index.html

[image]

It's just a wall of links. Something similar is in the Zephyr doc: https://docs.zephyrproject.org/latest/boards/index.html but in our case we have an additional sections with chip family which is valuable information but adds another level in the list.

Maybe a list with two columns would be more readable.