apache/nuttx #11087: Simulator on M1/M2 MacOS seems not working

duduita, 2023-10-29:
Simulator on M1/M2 MacOS seems not working:
```
Create version.h
Register: gpio
Register: hello
Register: dumpstack
Register: nsh
Register: sh
...
'g_reg_offs'
  .basic_num      = nitems(g_reg_offs),
                           ^
sim/sim_tcbinfo.c:96:28: error: use of undeclared identifier 'g_reg_offs'
sim/sim_tcbinfo.c:97:28: error: use of undeclared identifier 'g_reg_offs'
  .total_num      = nitems(g_reg_offs),
                           ^
sim/sim_tcbinfo.c:97:28: error: use of undeclared identifier 'g_reg_offs'
sim/sim_tcbinfo.c:99:10: error: use of undeclared identifier 'g_reg_offs'
    .p = g_reg_offs,
         ^
5 errors generated.
make[1]: *** [sim_tcbinfo.o] Error 1
make: *** [arch/sim/src/libarch.a] Error 2
```

Looking at the `sim_tcbinfo.c`, it seems that there are only `g_reg_offs` defined for `CONFIG_HOST_X86`, `CONFIG_HOST_X86_64`, and `CONFIG_SIM_M32`, but sounds that there's no `g_reg_offs` defined for `CONFIG_HOST_ARM64`.

xiaoxiang781216, 2023-10-29:
If you want to get better debugging experience (thread awareness) on macOS, you have to provide the register layout expected by gdb/clang:
https://github.com/apache/nuttx/blob/master/libs/libc/machine/sim/arch_setjmp_arm64.S
otherwise, you can justt provide a dummy g_reg_offs or skip g_tcbinfo.

duduita, 2023-10-29:
I tried to use a dummy `g_reg_offs`, but I got different errors
```
➜  nuttx git:(master) ✗ sudo make
LD:  nuttx
ld: warning: pointer not aligned at address 0x10006C612 ('_g_tcbinfo' + 18 from nuttx.rel)
ld: unaligned pointer(s) for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [nuttx] Error 1
make: *** [nuttx] Error 2
```

I don't want to use the debugger, I just want to be able to make the simulator on my mac. How can I skip the `g_tcbinfo`?

xiaoxiang781216, 2023-11-18:
@duduita could you verify whether https://github.com/apache/nuttx/pull/11207 can fix your issue?

duduita, 2023-11-19:
I'm currently vacationing, but I'll try it as soon as I can.

closed 2023-11-21 by xiaoxiang781216