apache/nuttx #11088: Reference design (demo) applications

raiden00pl, 2023-10-29:
Hi, I would like to propose another directory in nuttx-apps where we can place more complicated applications and demos (e.g., apps/demos, apps/refdesign) and in the future maybe even a separate repo if the idea works.

At the moment, we have many simple applications in `examples` but they are created more with the intention of showing individual features of the system. This is not very useful when testing on real hardware, because testing various features
requires running many individual apps often on different boards, which takes time, is irritating and boring.

Since not all users have the time to update their products with every new release, testing on real products is also typically significantly delayed.

"Demo" apps that touch on many aspects of NuttX can facilitate the testing process on a real HW and in the future, they may evolve into automatic testing on HW (integration tests).

One such "demo" app is already in examples, and it's `examples/foc` - basically a functional implementation of the FOC motor controller based on NuttX.

Soon, I would like to add an app that demonstrates NuttX on Thingy boards from Nordic (Thingy52, Thingy53 and Thingy91), with the main intention that it would work on other similar boards like SensorTile from ST.

There are many reference designs / evaluation boards from IC vendors that are good candidates for other demo applications. At the moment, I don't see a good place where you can place an application for such HW.

For testing purposes, we could keep a simple log for each such "demo" with additional information about when it was last tested, on what hardware and what features were verified.

The demo description could include a list of NuttX features that the app demonstrates and expected results.

Of course, applications of this type would have many more uses than testing. They would show NuttX features and portability, good practices with NuttX, provide a template for a specific group of problems, and more.

The downside of this approach is the additional code to be maintained by the community. On the other hand, any breaking change for this type of application would be a kind of guide for the user on how to fix their applications.

What is your opinion, does it make sense?

acassis, 2023-10-30:
@raiden00pl I think creating a name "demos" inside "apps/examples/" still keeping things confuse, because most examples are in fact demos already (demo how to use foc; demo how to use apds9960, etc).

It is better to create categories describing the type of example/demo.

Ie.:

Examples / Bluetooth
Examples / File Systems
Examples / Graphics
Examples / Industrial
Examples / Inertial
Examples / Interpreters
Examples / Network
Examples / Sensors

etc...

What do you think?

acassis, 2023-10-30:
Yes, I think we will need to have single categories like I suggested and some complex/jumbo demos/refdesign that include many functions.