apache/nuttx #11109: NuttX compile error

zouboan, 2023-11-01:
Hi:
I haven't tested NuttX in a long time. Today I updated my local branch to master branch, but i was surprised to find that the compilation failed. The compilation log is as follows:

`
zouboan@mini:~/d/FeedForward/nuttx$ tools/configure.sh stm32f4discovery:nsh
  Copy files
install: setting permissions for ‘/home/zouboan/d/FeedForward/nuttx/tools/../defconfig’: Operation not permitted
Failed to backup /home/zouboan/d/FeedForward/nuttx/tools/../boards/*/*/stm32f4discovery/configs/nsh/defconfig
zouboan@mini:~/d/FeedForward/nuttx$ 
zouboan@mini:~/d/FeedForward/nuttx$ 
zouboan@mini:~/d/FeedForward/nuttx$ sudo tools/configure.sh stm32f4discovery:nsh[sudo] password for zouboan: 
No configuration change.
zouboan@mini:~/d/FeedForward/nuttx$ 
zouboan@mini:~/d/FeedForward/nuttx$ 
zouboan@mini:~/d/FeedForward/nuttx$ make
sed: preserving permissions for ‘/home/zouboan/d/FeedForward/nuttx/sedkOIREX’: Operation not permitted
/usr/bin/bash: line 1: tools/mkconfig: No such file or directory
make: *** [tools/Unix.mk:254: include/nuttx/config.h] Error 127
zouboan@mini:~/d/FeedForward/nuttx$ 
zouboan@mini:~/d/FeedForward/nuttx$ sudo make
/usr/bin/bash: line 1: tools/mkconfig: No such file or directory
make: *** [tools/Unix.mk:254: include/nuttx/config.h] Error 127
zouboan@mini:~/d/FeedForward/nuttx$ 
zouboan@mini:~/d/FeedForward/nuttx$ 
zouboan@mini:~/d/FeedForward/nuttx$ make distclean
gcc: error: unrecognized command-line option ‘-mfloat-abi=soft’
gcc: error: unrecognized command-line option ‘-mfloat-abi=soft’
Makefile:22: chip/Make.defs: No such file or directory
make[1]: *** No rule to make target 'chip/Make.defs'.  Stop.
make: *** [tools/Unix.mk:748: arch/arm/src_clean] Error 2
`
Anyone who can give me some advice? thanks!

xiaoxiang781216, 2023-11-01:
why do you invoke configure.sh with sudo?

zouboan, 2023-11-01:
> why do you invoke configure.sh with sudo?

Because there's an error:Operation not permitted.
But a few months ago, there's no such error and it does not need sudo

xiaoxiang781216, 2023-11-02:
Please remove sudo from your command.

zouboan, 2023-11-02:
> Please remove sudo from your command.

zouboan@mini:~/d/FeedForward/nuttx$ tools/configure.sh stm32f4discovery:nsh
  Copy files
install: setting permissions for ‘/home/zouboan/d/FeedForward/nuttx/tools/../defconfig’: Operation not permitted
Failed to backup /home/zouboan/d/FeedForward/nuttx/tools/../boards/*/*/stm32f4discovery/configs/nsh/defconfig
zouboan@mini:~/d/FeedForward/nuttx$

zouboan, 2023-11-03:
The problem lies in the unexpected modification of nuttx directory permissions, thanks for your advice

closed 2023-11-03 by zouboan