apache/nuttx #11115: Problem with SMP and mm_delaylist

pussuw, 2023-11-02:
In SMP mode there is a mm_delaylist for each CPU. When calling mm_free() the memory is put into the delaylist if the mm_lock cannot be acquired, while the delaylist elements are freed when mm_malloc() is called.

There is a problem with the per-CPU delaylist when using this opportunistic approach. Let's imagine a hypothetical situation where:
- Only CPU 0 allocates memory
- Only CPU 1 frees memory 

This will cause a systematic error where CPU 1:s mm_delaylist is filled up with items to be freed, but since CPU 1 never calls mm_malloc() the items in the delaylist are never freed.

We can already see this behavior, as discussed in https://github.com/apache/nuttx-apps/pull/2159

In this simple example only 1 process (hello) is spawned and then the amount of free memory is queried from the shell. The kernel memory usage rises after each run of hello, due to the aforementioned issue.

This can and will eventually result in a OOM situation, even though most (if not all) of the memory is really free (waiting in CPU1 delaylist). Thus, the current opportunistic approach has a vulnerability.

TaiJuWu, 2023-11-11:
Current [mm_delay_list](https://github.com/apache/nuttx/blob/b6e09955ddae3484251c632501375ce6e5006295/mm/mm_heap/mm_free.c#L48) is protected by spinlock and only one CPU can operate mm_delay_list.

But our design is there is a mm_delay_list for each CPU, it will make this issue and delay_list for each CPU is unnecessary because only one CPU can operate delay_list .
Maybe we can share mm_delay_list for each heap instead of CPU?
If my idea is incorrect, please correct me.

Thanks a lot.

xiaoxiang781216, 2023-11-11:
no, we can't, you can get more info from: https://github.com/apache/nuttx/pull/4029

closed 2024-08-02 by xiaoxiang781216