apache/nuttx #11123: iob_free: io_offset not taken into consideration when adjusting remaining io_pktlen?

michi-jung, 2023-11-04:
Hello,

I am hitting this DEBUGASSERT occasionally with IP traffic on the SLIP network interface (Which due to a low MTU requires IP de-/fragmentation to work reasonably):

https://github.com/apache/nuttx/blob/c08ccbef029baedf4268756b5e221dae9a08ec65/mm/iob/iob_free.c#L100

While debugging I realized this happens in cases where the io_offset field of the IOB to be free'd is non-zero.  I believe the respective code should be changed as follows:

```
diff --git a/mm/iob/iob_free.c b/mm/iob/iob_free.c
index 232e28ff4e..715c062b81 100644
--- a/mm/iob/iob_free.c
+++ b/mm/iob/iob_free.c
@@ -92,11 +92,11 @@ FAR struct iob_s *iob_free(FAR struct iob_s *iob)
        * do nothing too crazy.
        */
 
-      if (iob->io_pktlen > iob->io_len)
+      if (iob->io_pktlen > (iob->io_len - iob->io_offset))
         {
           /* Adjust packet length and move it to the next entry */
 
-          next->io_pktlen = iob->io_pktlen - iob->io_len;
+          next->io_pktlen = iob->io_pktlen - (iob->io_len - iob->io_offset);
           DEBUGASSERT(next->io_pktlen >= next->io_len);
         }
       else
```

Could somebody with a good understanding of IOB handling have a look at this, please?

Thanks!
Michael

xiaoxiang781216, 2023-11-05:
@wengzhe could you reply @michi-jung 's question?

wengzhe, 2023-11-05:
As far as I know, both `io_offset` and `io_len` do not count `io_offset`, which means the sum of `io_len` should be `io_offset`. If not, I believe it's the operation that makes them unequal gets wrong.

You can refer to iob_test.c which always checks whether they're equal. (I don't know the official way to run this test, I put it into apps and ran it as an app previously)
https://github.com/apache/nuttx/blob/c08ccbef029baedf4268756b5e221dae9a08ec65/mm/iob/iob_test.c#L62-L75

michi-jung, 2023-11-05:
@wengzhe Thanks!  I am going to investigate further then.

michi-jung, 2023-11-05:
```
(gdb) print *(struct iob_s *)0x2000fdec
$14 = {io_flink = 0x2000fd1c <g_iob_buffer+16224>, io_len = 182 '\266', io_offset = 14 '\016', io_pktlen = 187,
  io_data = "\006T21\003T111.111\034E\000\000\311A\003@\000\001\021>&\n\000\000\001\357\377\377\372\276\336\al\000\265K-M-SEARCH * HTTP/1.1\r\nHOST: 239.255.255.250:1900\r\nMAN: \"ssdp:discover\"\r\nMX: 1\r\nST: urn:dial-multiscreen-org:service:dial:1\r\nUSER-AGENT: Google Chrome/119.0"}
(gdb) print *(struct iob_s *)0x2000fd1c
$15 = {io_flink = 0x0, io_len = 19 '\023', io_offset = 0 '\000', io_pktlen = 5,
  io_data = ".6045.105 Linux\r\n\r\n\031@\000@\006&\265\n\000\000\002\n\000\000\001&\224\240\332;\232\314\035\277\300i\351P\020\066[l\246\000\000\002T21\003T \n\036f\245/\000\000\000\000\001\003\003\a", '\000' <repeats 121 times>}

450000C9 IPv4 - 5 Word Header Len - 201 byte total len
41034000 Identification (0x4103) - DF (Don't fragment)
01113E26 Time to Live 0x01 - Protocol 0x11 (UDP) - Header Checksum 0x3E26
0D000001 Source IP Address 13.0.0.1
EFFFFFFA Destination IP Address 239.255.255.250 (Multicast address used in UPnP)

BEDE076C Source port 48862, Destination port 1900 (UPnP)
00B54B2D Length 181 - Checksum 0x4B2D

Payload (154 bytes)
M-SEARCH * HTTP/1.1\r\nHOST: 239.255.255.250:1900\r\nMAN: \"ssdp:discover\"\r\nMX: 1\r\nST: urn:dial-multiscreen-org:service:dial:1\r\nUSER-AGENT: Google Chrome/119.0

Payload (19 bytes)
.6045.105 Linux\r\n\r\n\
```
The io_pktlen field is to low by 14 bytes in both IOBs (which equals the io_offset of the first IOB of the chain, which is CONFIG_NET_LL_GUARDSIZE=14)

wengzhe, 2023-11-05:
Are you working under code after commit e2c9aa6? That commit fixes a similar problem when applying `iob_reserve` on an IOB chain twice.

If you're working with the newest code, maybe there's a problem in the fragmentation process or some other place, the chain is weird because normally the second IOB's `io_pktlen` would be 0 (both `iob_copyin` and `iob_concat` will not make it non-zero).

michi-jung, 2023-11-05:
Thanks again for your help, @wengzhe 

Indeed, I am missing this fix (I am still on NuttX 12.0.0).  Meanwhile I cherrry-picked it and it did resolve my problem.

closed 2023-11-05 by michi-jung