apache/nuttx #11127: Documentation: migrate pages from wiki

raiden00pl, 2023-11-04:
Not all pages below are suitable for migration, some pages are outdated, some are already migrated. Everything listed here for better tracking of the issue.

**Getting Started with NuttX** https://cwiki.apache.org/confluence/display/NUTTX/Getting+Started+with+NuttX

- [X] nothing to migrate

**Graphics** https://cwiki.apache.org/confluence/display/NUTTX/Graphics
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Framebuffer+Character+Driver
- [X] https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629474
- [X] https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139630154
- [X] https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629473
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/CNxConsole+Keyboard+Input
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/NxWM+Threading
- [X] https://cwiki.apache.org/confluence/display/NUTTX/Wide+Font+Support
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/NxTerm+Example

**NuttX How-To's** https://cwiki.apache.org/confluence/display/NUTTX/NuttX+How-To%27s
  
- [X] https://cwiki.apache.org/confluence/display/NUTTX/Analyzing+Cortex-M+Hardfaults
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/ARMv7-M+Run+Time+Stack+Checking
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Auto-Mounter
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Changing+the+System+Clock+Configuration
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Converting+Legacy+Configurations+to+Use+kconfig-mconf[]
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Converting+Legacy+Configurations+to+Use+kconfig-mconf
- [ ] https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629444
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Disabling+the+Stack+Dump+During+Debugging
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Including+Files+in+board.h
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Integrating+with+Newlib
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/IPv6
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Kernel+Threads+with+Custom+Stacks
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Logging+to+a+RAM+Buffer
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/NuttX+Protected+Build
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Platform+Directories
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Porting+Drivers+to+the+STM32+F7
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Semihosting
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Signaling+Events+from+Interrupt+Handlers
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Signaling+Semaphores+and+Priority+Inheritance
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Smaller+Vector+Tables
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/STM32+CCM+Allocator
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/STM32+Null+Pointer+Detection
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Using+Kernel+Threads
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Versioning+and+Task+Names
- [ ] https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629466

**Networking** https://cwiki.apache.org/confluence/display/NUTTX/Networking

- [ ] https://cwiki.apache.org/confluence/display/NUTTX/CONFIG_NET_GUARDSIZE
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Delayed+ACK+and+TCP+Performance
- [X] https://cwiki.apache.org/confluence/display/NUTTX/Network+Link+Management
   - nothing to migrate
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/SLIP+Configuration
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/TCP+Network+Performance
- [ ] https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629530
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Work+Queue+Deadlocks

**Application/NSH How-To's** https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629534

- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Updating+a+Release+System+with+ELF+Programs
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/APPNAME+vs.+PROGNAME
- [ ] https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629538
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Building+NuttX+with+Applications+Outside+of+the+Source+Tree
- [ ] https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629550
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Custom+Application+Directories
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Debugging+ELF+Loadable+Modules
- [ ] https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629545
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Multiple+NSH+Sessions
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/NSH+Network+Link+Management
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/RAM+Disks+and+ROM+Disks
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Reading+CAN+Messages
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Removing+Device+Drivers+with+NSH
[... 31 lines cut ...]
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Short+Time+Delays
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Signal+Handler+Tour
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/SMP
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/SMP+Critical+Sections
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/SYSLOG
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Task+Exit+Sequence
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Tasks+vs.+Threads+FAQ
- [ ] https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=152115079
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Tickless+OS
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Upper+Half+and+Lower+Half+Drivers
- [ ] https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=158862687

**Platforms** https://cwiki.apache.org/confluence/display/NUTTX/Platforms

- [X] nothing to migrate

**Virtual File System** https://cwiki.apache.org/confluence/display/NUTTX/Virtual+File+System

- [ ] https://cwiki.apache.org/confluence/display/NUTTX/How+NXFFS+Works
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/NuttX+File+System
- [ ] https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139630111
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Pseudo+File+System
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/SmartFS+Internals
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Special+Files+and+Device+Numbers
- [ ] https://cwiki.apache.org/confluence/display/NUTTX/Using+SmartFS

lvanasse, 2024-09-29:
@raiden00pl would it still be useful to work on the documentation even if it is outdated? I understand that there's not a lot of reason to work on getting outdated documentation in the current documentation format, but I'm thinking it might be difficult to see what is outdated or not. Also, you might need setups to validate how outdated are the pages.

What are your thoughts?

raiden00pl, 2024-09-30:
Unless the doc page is completely outdated, porting it still has value, and outdated content can be fixed later.
We can mark such pages with `.. warning::` note at the top until they are fixed.

As you mentioned, it can be difficult to determine what is outdated and what isn't, especially for people who haven't been following all the PRs in NuttX over the last few years.

So I'm not sure how we should proceed. Maybe we can just hope that during the PR review process, we'll be able to catch any outdated content and fix it immediately or mark the page as outdated.

lvanasse, 2024-10-14:
@raiden00pl This page: [https://cwiki.apache.org/confluence/display/NUTTX/STM32+Null+Pointer+Detection](https://cwiki.apache.org/confluence/display/NUTTX/STM32+Null+Pointer+Detection) was already migrated :). #14231 Noticed just too late :sweat_smile:

lvanasse, 2024-10-14:
@raiden00pl darn again, very sorry, just notice: [https://cwiki.apache.org/confluence/display/NUTTX/STM32+CCM+Allocator](https://cwiki.apache.org/confluence/display/NUTTX/STM32+CCM+Allocator) also was already migrated :sweat_smile:

lvanasse, 2024-10-14:
@raiden00pl here are 3 other pages to check (once they are merged) :) 

#14240 - Kernel Threads with Custom Stacks
#14239 - Versioning and Task Names
#14236 - Using Kernel Threads 

Cheers!

lvanasse, 2024-10-14:
@raiden00pl here are 3 other pages to check (once they are merged) :)

#14274 : 
- Disabling the Stack Dump During Debugging
- Including files in board.
- Why can't I put my special stuff in NuttX header files?

Cheers!

lvanasse, 2024-10-20:
@raiden00pl here are 4 other pages to check (once they are merged) :)

#14417 : 
* NuttX Protected Build
* Integrating with Newlib
* IPv6
* Logging to a RAM buffer

Cheers!

raiden00pl, 2024-10-21:
great ! thanks for your help :)

jerpelea, 2024-10-21:
@raiden00pl thanks for doing it

lvanasse, 2024-10-27:
@raiden00pl here are 3 other pages to check (once they are merged) :)

https://github.com/apache/nuttx/pull/14522:
* Signaling Events from Interrupt Handlers
* Signaling Semaphores and priority inheritance
* Smaller Vector Tables

(NuttX How-To's done :partying_face:)

Cheers!

lvanasse, 2024-11-09:
@raiden00pl here's 4 other pages to check (once they are merged) :)

#14710 : 
* Delayed ACK and TCP Performance
* TCP Network Performance
* NxWM Threading
* Framebuffer Character Driver

(Graphics and Networking section done! Only two more sections to go)

Cheers!

lvanasse, 2024-11-10:
@raiden00pl, please correct me if I'm wrong, but I think we are missing this page in the list of document to port:

https://cwiki.apache.org/confluence/display/NUTTX/Porting+Guide

I did check in the official wiki, and didn't find it there, and I'm not seeing it in the list when I `ctrl+f` for it.

lvanasse, 2024-11-10:
@raiden00pl here are 4 other pages to check (once they are merged) :)

https://github.com/apache/nuttx/pull/14714 :

* [Special Files and Device Numbers](https://cwiki.apache.org/confluence/display/NUTTX/Special+Files+and+Device+Numbers)
* [Pseudo File System](https://cwiki.apache.org/confluence/display/NUTTX/Pseudo+File+System)
* [How NXFFS Works](https://cwiki.apache.org/confluence/display/NUTTX/How+NXFFS+Works)
* https://cwiki.apache.org/confluence/display/NUTTX/NuttX+File+System
* https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139630111

Cheers!

raiden00pl, 2024-11-11:
@lvanasse 

Doc for "On-Demand Paging" is here https://nuttx.apache.org/docs/latest/components/paging.html
"Porting Guide" and "NuttX Overview" are most likely outdated and I don't know if it makes sense to port them without a proper review.
"User Guide" is probably missing new features, but it should be OK to add.

lvanasse, 2024-12-28:
@raiden00pl I see that you've written about the SMARTFS in the official documentation. I was wondering if we ought to still do the
* https://cwiki.apache.org/confluence/display/NUTTX/Using+SmartFS
* https://cwiki.apache.org/confluence/display/NUTTX/SmartFS+Internals

pages?

lvanasse, 2024-12-29:
@raiden00pl here's 1 other page to check (once they are merged) :)

https://github.com/apache/nuttx/pull/15376 :

* [Updating a Release System with ELF Programs](https://cwiki.apache.org/confluence/display/NUTTX/Updating+a+Release+System+with+ELF+Programs)
* [ELF Programs – With Symbol Tables](https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629543)
* [ELF Programs – No Symbol Tables](https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629542)

The two ELF Programs pages weren't in the list, but I couldn't find them inside the official wiki.

Cheers!

lvanasse, 2024-12-29:
Another question for @raiden00pl or maybe @acassis since it came from you. It's regarding the page: https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629545

It refers to the libc++ of LLVM, but I'm unsure how relevant this page is since we now have the https://nuttx.apache.org/docs/latest/components/libs/libxx.html page inside the official documentation. From reading both it seems that the Confluence page kind of share the repo @acassis made to use the LLVM libc++, and tell me if I'm wrong, but we don't seem to use those actively? It seems to be integrated into the KConfig with `CONFIG_LIBCXX=y` as per the official documentation is saying. 

Anyway, I don't mind migrating it, just didn't want to do redundant work :) If you could please let me know if you still want this page migrated, that would be appreciated :).

lvanasse, 2024-12-30:
@raiden00pl here are 9 other pages to check, once merged :)

https://github.com/apache/nuttx/pull/15379 

* Building NuttX with Applications Outside the Source Tree
* Building uClibc++
* Custom Application Directories
* Debugging ELF Loadable Modules
* Multiple NSH Sessions
* NSH Network Link Management
* RAM Disks and ROM Disks
* Reading CAN Messages
* Removing Device Drivers with NSH

I've also stumbled upon the https://cwiki.apache.org/confluence/pages/viewpage.action?pageId=139629410 I was wondering if it should also be added to the list, it seems very interesting.

Also, that PR concluded the NSH How-To's :)!

Cheers!

acassis, 2024-12-31:
@lvanasse that page is very old and as you noticed LLVM C++ is already integrated into mainline. I suggest removing it or creating a new page explaining how to use LLVM C++ nowadays.

lvanasse, 2025-01-04:
@acassis I'm not too sure what would be the content of how to use LLVM C++ nowadays page to be honest :disappointed_relieved:, so I'm hesitant to make a call.

acassis, 2025-01-05:
@lvanasse yes, since the integration is transparent now, there is not much to document

lvanasse, 2025-01-05:
> @raiden00pl I see that you've written about the SMARTFS in the official documentation. I was wondering if we should to still do the
> 
>     * https://cwiki.apache.org/confluence/display/NUTTX/Using+SmartFS
> 
>     * https://cwiki.apache.org/confluence/display/NUTTX/SmartFS+Internals
> 
> 
> pages? And if yes, should we put them as supporting pages to the one you've written?

@raiden00pl any input :)?