apache/nuttx #11159: Failure in compilation of sim_hostusrsock.c due to improper handling of header inclusion for fd_set

PwnVerse, 2023-11-09:
Compilation failure may result due to missing header which defines the fd_set structure in [sim_hostusrsock.c](https://github.com/apache/nuttx/blob/master/arch/sim/src/sim/posix/sim_hostusrsock.c#L49-L50).

```
/home/ritvik/Desktop/Files/Fuzzing_RTOS_Applications/Ported_Applications/nuttxspace/nuttx/arch/sim/src/sim/posix/sim_hostusrsock.c:51:8: error: unknown type name 'fd_set'
   51 | static fd_set g_active_read_fds;
      |        ^
/home/ritvik/Desktop/Files/Fuzzing_RTOS_Applications/Ported_Applications/nuttxspace/nuttx/arch/sim/src/sim/posix/sim_hostusrsock.c:52:8: error: unknown type name 'fd_set'
   52 | static fd_set g_active_write_fds;
```

fd_set is only defined in [select.h](https://github.com/apache/nuttx/blob/c655870b22736e53982efdd4f8c8f8c919ed8320/include/sys/select.h#L106) which should be included in this header or any of the subsequent headers included in this file.

xiaoxiang781216, 2023-11-09:
could you provide a patch to fix.

acassis, 2023-11-09:
@PwnVerse very nice! I hope you help to find issue on NuttX fuzzing it! ;-)