apache/nuttx #11163: Failed to build simulator

mildsky, 2023-11-10:
I tried to build simulator on my M2 MBA, It produce error like below.
`
➞  ./tools/configure.sh -m sim:nsh
No configuration change.
➞  make clean
➞  make -j8
Create version.h
CC:  misc/lib_fnmatch.c sched/sched_cpuload.c:109:25: error: use of undeclared identifier 'CONFIG_SCHED_CPULOAD_TIMECONSTANT'
  if (g_cpuload_total > CPULOAD_TIMECONSTANT)
                        ^
sched/sched_cpuload.c:59:7: note: expanded from macro 'CPULOAD_TIMECONSTANT'
      CONFIG_SCHED_CPULOAD_TIMECONSTANT * \
      ^
1 error generated.
make[1]: *** [sched_cpuload.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [sched/libsched.a] Error 2
make: *** Waiting for unfinished jobs....
`
I fetched master branch before I build.

acassis, 2023-11-10:
@mildsky  there is something wrong on your config! sched_cpuload.c is only compiled when CONFIG_SCHED_CPULOAD is enabled and by default it is disabled:
```
$ cat .config | grep CONFIG_SCHED_CPULOAD
# CONFIG_SCHED_CPULOAD is not set
```
Did you run "make menuconfig" and added it manually? Anyway is it strange because when I enable CONFIG_SCHED_CPULOAD into menuconfig, it automatically defines that symbol that is missing on your setup:

```
$ cat .config | grep CONFIG_SCHED_CPULOAD_TIMECONSTANT
CONFIG_SCHED_CPULOAD_TIMECONSTANT=2
```

BTW, I'm using Linux, not sure if it could be something specific to MacOS M2

mildsky, 2023-11-11:
I didn't add manually, but after I tried to add it, `CONFIG_SCHED_CPULOAD_TIMECONSTANT` was properly setted as 2. But, there are still some bugs that I can't understand.
here are my console output.
```
$ make clean

$ ./tools/configure.sh -m sim:nsh                     
No configuration change.

$ cat .config | grep CONFIG_SCHED_CPULOAD             
# CONFIG_SCHED_CPULOAD_NONE is not set
CONFIG_SCHED_CPULOAD_SYSCLK=y
# CONFIG_SCHED_CPULOAD_EXTCLK is not set
CONFIG_SCHED_CPULOAD_TIMECONSTANT=2

$ cat .config | grep CONFIG_SCHED_CPULOAD_TIMECONSTANT
CONFIG_SCHED_CPULOAD_TIMECONSTANT=2

$ make -j8  
LD:  nuttx
ld: Undefined symbols:
  _nxsched_process_cpuload_ticks, referenced from:
      _main in nuttx.rel
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[1]: *** [nuttx] Error 1
make: *** [nuttx] Error 2
```
I think that simulator not yet supported on apple silicon machine.
Thank you.