apache/nuttx #11223: gdb nsh:tcpblaster cannot access network

maxmbed, 2023-11-19:
When running Nuttx `sim:tcpblaster` in GDB , I am encountering a problem to access the network. 
I cannot seem to access or to ping either the gateway or dns 8.8.8.8.

I tried to set kernel capabilities of gdb to access the tap device but no success
```
sudo setcap cap_net_admin+ep /usr/bin/gdb
``` 
Nuttx version: master branch b5d640acc

Here is some logs wit hNetwork Debug Features enabled.

Ping failure:
```
nsh> ping -c 1 10.0.1.1
PING 10.0.1.1 56 bytes of data
[42949807.110000] arp_send_eventhandler: flags: 2000 sent: 0
[42949807.140000] arp_send: ERROR: arp_wait failed: -110, ipaddr: 10.0.1.1
[42949807.140000] arp_send_eventhandler: flags: 2000 sent: 0
[42949807.170000] arp_send: ERROR: arp_wait failed: -110, ipaddr: 10.0.1.1
[42949807.170000] arp_send_eventhandler: flags: 2000 sent: 0
[42949807.200000] arp_send: ERROR: arp_wait failed: -110, ipaddr: 10.0.1.1
[42949807.200000] arp_send_eventhandler: flags: 2000 sent: 0
[42949807.230000] arp_send: ERROR: arp_wait failed: -110, ipaddr: 10.0.1.1
[42949807.230000] arp_send_eventhandler: flags: 2000 sent: 0
[42949807.260000] arp_send: ERROR: arp_wait failed: -110, ipaddr: 10.0.1.1
[42949807.260000] icmp_sendmsg: ERROR: Not reachable
ERROR: sendto failed at seqno 0: 101
nsh>
nsh> ping -c 1 8.8.8.8
PING 8.8.8.8 56 bytes of data
[42951305.590000] arp_send_eventhandler: flags: 2000 sent: 0
[42951305.620000] arp_send: ERROR: arp_wait failed: -110, ipaddr: 10.0.1.1
[42951305.620000] arp_send_eventhandler: flags: 2000 sent: 0
[42951305.650000] arp_send: ERROR: arp_wait failed: -110, ipaddr: 10.0.1.1
[42951305.650000] arp_send_eventhandler: flags: 2000 sent: 0
[42951305.680000] arp_send: ERROR: arp_wait failed: -110, ipaddr: 10.0.1.1
[42951305.680000] arp_send_eventhandler: flags: 2000 sent: 0
[42951305.710000] arp_send: ERROR: arp_wait failed: -110, ipaddr: 10.0.1.1
[42951305.710000] arp_send_eventhandler: flags: 2000 sent: 0
[42951305.740000] arp_send: ERROR: arp_wait failed: -110, ipaddr: 10.0.1.1
[42951305.740000] icmp_sendmsg: ERROR: Not reachable
ERROR: sendto failed at seqno 0: 101

```

Ping success:

```
nsh> ping -c 1 10.0.1.2
PING 10.0.1.2 56 bytes of data
[42951267.190000] sendto_eventhandler: flags: 4000
[42951267.190000] sendto_eventhandler: Send ICMP request
[42951267.190000] ipv4_build_header: IPv4 Packet: ipid:12, length: 84
[42951267.190000] sendto_request: Outgoing ICMP packet length: 84
[42951267.190000] sendto_eventhandler: Resuming
[42951267.190000] devif_loopback: IPv4 frame
[42951267.190000] icmp_input: Outgoing ICMP packet length: 84 (84)
[42951267.190000] devif_loopback: IPv4 frame
[42951267.190000] icmp_datahandler: Buffered 84 bytes
[42951267.190000] icmp_readahead: Received 64 bytes (of 84)
56 bytes from 10.0.1.2: icmp_seq=0 time=0.0 ms
1 packets transmitted, 1 received, 0% packet loss, time 1010 ms
rtt min/avg/max/mdev = 0.000/0.000/0.000/0.000 ms

```

IP config of Nuttx sim:
```
nsh> ifconfig
[42949813.100000] netdev_ifr_ioctl: cmd: 1819
[42949813.100000] netdev_ifr_ioctl: cmd: 1819
[42949813.100000] netdev_ifr_ioctl: cmd: 1819
[42949813.100000] netdev_ifr_ioctl: cmd: 1819
eth0    Link encap:Ethernet HWaddr 00:00:00:00:00:00 at RUNNING mtu 1280
        inet addr:10.0.1.2 DRaddr:10.0.1.1 Mask:255.255.255.0
        inet6 addr: fc00::2/112
        inet6 DRaddr: fc00::1/112

        RX: Received Fragment Errors
            00000000 00000000 00000000
            IPv4     IPv6     ARP      Dropped
            00000000 00000000 00000000 00000000
        TX: Queued   Sent     Errors   Timeouts
            0000000f 00000000 00000000 00000000
        Total Errors: 00000000

[42949813.100000] netdev_ifr_ioctl: cmd: 1819
lo      Link encap:Local Loopback at RUNNING mtu 1518
        inet addr:127.0.0.1 DRaddr:127.0.0.1 Mask:255.0.0.0
        inet6 addr: ::1/128
        inet6 DRaddr: ::1/128

        RX: Received Fragment Errors
            00000000 00000000 00000000
            IPv4     IPv6     ARP      Dropped
            00000000 00000000 00000000 00000000
        TX: Queued   Sent     Errors   Timeouts
            00000000 00000000 00000000 00000000
        Total Errors: 00000000

             IPv4  IPv6   TCP   UDP  ICMP  ICMPv6
Received     0000  0000  0000  0000  0000  0000
Dropped      0000  0000  0000  0000  0000  0000
  IPv4        VHL: 0000   Frg: 0000
  IPv6        VHL: 0000
  Checksum   0000  ----  0000  0000  ----  ----
  TCP         ACK: 0000   SYN: 0000
              RST: 0000  0000
  Type       0000  0000  ----  ----  0000  0000
Sent         0000  0000  0000  0000  0000  0000
  Rexmit     ----  ----  0000  ----  ----  ----
```

Host tap ip config:
```
18: tap0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 3a:cb:fe:a5:47:53 brd ff:ff:ff:ff:ff:ff
    inet 10.0.1.1/24 scope global tap0
       valid_lft forever preferred_lft forever
    inet6 fc00::1/112 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::5497:73ff:fe61:d490/64 scope link proto kernel_ll
       valid_lft forever preferred_lft forever
```

alandeassis, 2023-11-20:
@maxmbed can you ping NuttX from your host? You need ICMP enabled on NuttX!

maxmbed, 2023-11-21:
@alandeassis, No, the host cannot ping Nuttx in this case.

Actually, the `sim:tcpblaster` example has already ICMP support enabled. 
Network is accessible if I run tcpblaster example and follow [this guide](https://nuttx.apache.org/docs/latest/guides/simulator.html#accessing-the-network) but when it is launched with GDB, network access does not work.

`gdb ./build/nuttx tools/gdb/__init__.py`

acassis, 2023-11-22:
@masayuki2009 any idea why network in the SIM is not accessing when using GDB ?

anjiahao1, 2023-11-23:
> @alandeassis, No, the host cannot ping Nuttx in this case.
>
> Actually, the `sim:tcpblaster` example has already ICMP support enabled. Network is accessible if I run tcpblaster example and follow [...]

you mean just run `nuttx` network is ok？ but `nuttx` run with gdb, network get error ？

anjiahao1, 2023-11-23:
[image]
@maxmbed It may be caused by gdb not having network permissions? Try using sudo gdb

maxmbed, 2023-11-25:
Thanks @anjiahao1, launching gdb with the admin privilege is working. I can access network now.

Without sudo, the issue happens during the configuration of the tap device in `nuttx/arch/sim/src/sim/posix/sim_tapdev.c`
A error message is logged to host syslog but it is not prompted in the Nuttx shell.    

```c
 207    /* Configure the tap device */
 208
 209    memset(&ifr, 0, sizeof(ifr));
 210    ifr.ifr_flags = IFF_TAP | IFF_NO_PI;
 211    ret = ioctl(tapdevfd, TUNSETIFF, (unsigned long) &ifr);
 212    if (ret < 0)
 213      {
 214        syslog(LOG_ERR, "TAPDEV: ioctl failed: %d\n", -ret);
 215        close(tapdevfd);
 216        return;
 217      }
```
Though, I don't like the idea to give admin privilege to GDB because there is good reason to use theLinux capabilities but that does not seems to work when I try  `sudo setcap cap_net_admin+ep /usr/bin/gdb` or `setcap cap_net_raw,cap_net_admin=eip /usr/bin/gdb`.

maxmbed, 2023-12-05:
Cannot find a way to apply Linux capability to gdb. If anyone has solution, please let me us know here.