apache/nuttx #11227: stringstream linker problem when Normal, Full Optimization Activated

trns1997, 2023-11-20:
# Error
```
P:➜  nuttx git:(master) ✗ make -j8       
LD: nuttx
arm-none-eabi-ld: warning: /home/trns1997/nuttxspace/nuttx/nuttx has a LOAD segment with RWX permissions
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `__tcf_0':
locale.cpp:(.text.__tcf_0+0x10): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `__tcf_1':
locale.cpp:(.text.__tcf_1+0x10): undefined reference to `std::__1::basic_string<wchar_t, std::__1::char_traits<wchar_t>, std::__1::allocator<wchar_t> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `__tcf_2':
locale.cpp:(.text.__tcf_2+0x10): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `__tcf_3':
locale.cpp:(.text.__tcf_3+0x10): undefined reference to `std::__1::basic_string<wchar_t, std::__1::char_traits<wchar_t>, std::__1::allocator<wchar_t> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `__tcf_4':
locale.cpp:(.text.__tcf_4+0x6): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
arm-none-eabi-ld: locale.cpp:(.text.__tcf_4+0x12): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `__tcf_5':
locale.cpp:(.text.__tcf_5+0x6): undefined reference to `std::__1::basic_string<wchar_t, std::__1::char_traits<wchar_t>, std::__1::allocator<wchar_t> >::~basic_string()'
arm-none-eabi-ld: locale.cpp:(.text.__tcf_5+0x12): undefined reference to `std::__1::basic_string<wchar_t, std::__1::char_traits<wchar_t>, std::__1::allocator<wchar_t> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::numpunct<char>::~numpunct()':
locale.cpp:(.text._ZNSt3__18numpunctIcED2Ev+0xa): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::numpunct<wchar_t>::~numpunct()':
locale.cpp:(.text._ZNSt3__18numpunctIwED2Ev+0xa): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::__time_get_c_storage<char>::__x() const':
locale.cpp:(.text._ZNKSt3__120__time_get_c_storageIcE3__xEv+0x44): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::__time_get_c_storage<char>::__X() const':
locale.cpp:(.text._ZNKSt3__120__time_get_c_storageIcE3__XEv+0x44): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::__time_get_c_storage<char>::__c() const':
locale.cpp:(.text._ZNKSt3__120__time_get_c_storageIcE3__cEv+0x44): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o):locale.cpp:(.text._ZNKSt3__120__time_get_c_storageIcE3__rEv+0x44): more undefined references to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()' follow
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::__time_get_c_storage<wchar_t>::__x() const':
locale.cpp:(.text._ZNKSt3__120__time_get_c_storageIwE3__xEv+0x4c): undefined reference to `std::__1::basic_string<wchar_t, std::__1::char_traits<wchar_t>, std::__1::allocator<wchar_t> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::__time_get_c_storage<wchar_t>::__X() const':
locale.cpp:(.text._ZNKSt3__120__time_get_c_storageIwE3__XEv+0x4c): undefined reference to `std::__1::basic_string<wchar_t, std::__1::char_traits<wchar_t>, std::__1::allocator<wchar_t> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::__time_get_c_storage<wchar_t>::__c() const':
locale.cpp:(.text._ZNKSt3__120__time_get_c_storageIwE3__cEv+0x4c): undefined reference to `std::__1::basic_string<wchar_t, std::__1::char_traits<wchar_t>, std::__1::allocator<wchar_t> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::__time_get_c_storage<wchar_t>::__r() const':
locale.cpp:(.text._ZNKSt3__120__time_get_c_storageIwE3__rEv+0x4c): undefined reference to `std::__1::basic_string<wchar_t, std::__1::char_traits<wchar_t>, std::__1::allocator<wchar_t> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::num_put<char, std::__1::ostreambuf_iterator<char, std::__1::char_traits<char> > >::do_put(std::__1::ostreambuf_iterator<char, std::__1::char_traits<char> >, std::__1::ios_base&, char, bool) const':
locale.cpp:(.text._ZNKSt3__17num_putIcNS_19ostreambuf_iteratorIcNS_11char_traitsIcEEEEE6do_putES4_RNS_8ios_baseEcb[_ZNKSt3__17num_putIcNS_19ostreambuf_iteratorIcNS_11char_traitsIcEEEEE6do_putES4_RNS_8ios_baseEcb]+0x70): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::num_put<wchar_t, std::__1::ostreambuf_iterator<wchar_t, std::__1::char_traits<wchar_t> > >::do_put(std::__1::ostreambuf_iterator<wchar_t, std::__1::char_traits<wchar_t> >, std::__1::ios_base&, wchar_t, bool) const':
locale.cpp:(.text._ZNKSt3__17num_putIwNS_19ostreambuf_iteratorIwNS_11char_traitsIwEEEEE6do_putES4_RNS_8ios_baseEwb[_ZNKSt3__17num_putIwNS_19ostreambuf_iteratorIwNS_11char_traitsIwEEEEE6do_putES4_RNS_8ios_baseEwb]+0x6e): undefined reference to `std::__1::basic_string<wchar_t, std::__1::char_traits<wchar_t>, std::__1::allocator<wchar_t> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::num_get<char, std::__1::istreambuf_iterator<char, std::__1::char_traits<char> > >::do_get(std::__1::istreambuf_iterator<char, std::__1::char_traits<char> >, std::__1::istreambuf_iterator<char, std::__1::char_traits<char> >, std::__1::ios_base&, unsigned int&, void*&) const':
locale.cpp:(.text._ZNKSt3__17num_getIcNS_19istreambuf_iteratorIcNS_11char_traitsIcEEEEE6do_getES4_S4_RNS_8ios_baseERjRPv[_ZNKSt3__17num_getIcNS_19istreambuf_iteratorIcNS_11char_traitsIcEEEEE6do_getES4_S4_RNS_8ios_baseERjRPv]+0x17e): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
arm-none-eabi-ld: locale.cpp:(.text._ZNKSt3__17num_getIcNS_19istreambuf_iteratorIcNS_11char_traitsIcEEEEE6do_getES4_S4_RNS_8ios_baseERjRPv[_ZNKSt3__17num_getIcNS_19istreambuf_iteratorIcNS_11char_traitsIcEEEEE6do_getES4_S4_RNS_8ios_baseERjRPv]+0x184): undefined reference to `std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::~basic_string()'
arm-none-eabi-ld: /home/trns1997/nuttxspace/nuttx/staging/libxx.a(locale.o): in function `std::__1::num_get<wchar_t, std::__1::istreambuf_iterator<wchar_t, std::__1::char_traits<wchar_t> > >::do_get(std::__1::istreambuf_iterator<wchar_t, std::__1::char_traits<wchar_t> >, std::__1::istreambuf_iterator<wchar_t, std::__1::char_traits<wchar_t> >, std::__1::ios_base&, unsigned int&, void*&) const':
locale.cpp:(.text._ZNKSt3__17num_getIwNS_19istreambuf_iteratorIwNS_11char_traitsIwEEEEE6do_getES4_S4_RNS_8ios_baseERjRPv[_ZNKSt3__17num_getIwNS_19istreambuf_iteratorIwNS_11char_traitsIwEEEEE6do_getES4_S4_RNS_8ios_baseERjRPv]+0x18c) [... cut ...]
5. gcc version: arm-none-eabi-g++ (Arm GNU Toolchain 12.3.Rel1 (Build arm-12.35)) 12.3.1 20230626
6. It is not board specific it occurs everytime, if we `Suppress Optimization`  in `Build Setup` then the bin is compiled (in the case of stm32f411e-disco it fails simply because the bin produced is larger than the flash).
7. I tried commits/tags before this PR https://github.com/apache/nuttx/pull/8244 and the binary builds with optimization. `gcc version: arm-none-eabi-g++ (15:10.3-2021.07-4) 10.3.1 20210621 (release)` and `libcxx version: 12.0.0`

acassis, 2023-11-24:
@xiaoxiang781216 any idea why C++ is failing?

xiaoxiang781216, 2023-11-25:
not sure why.

leducp, 2023-11-27:
BTW, we build the last LLVM version (17.0.5) but the problem is the same.

trns1997, 2023-11-27:
@xiaoxiang781216 the problem specifically arises when we activate the `-Os/1/2/3` flags while compiling.

SPRESENSE, 2024-03-12:
@trns1997 @xiaoxiang781216 
I have the same problem with the Spresense board. What is the status of this issue?

trns1997, 2024-03-12:
@SPRESENSE ah sadly it is not as trivial as we thought. @leducp found an alternative approach to string stream with char conv, he could give you some tips and guide you through it. We might look into this further but i would not bet on it anytime soon. Though i would consider rebasing and trying the following: https://github.com/apache/nuttx/pull/11812 upgrade we added, where we use the std c++ lib provided by the toolchain, with any luck it resolves your issue. Do not hesitate to reach out if you test out the above approach.