apache/nuttx #11236: Fail to nxtask_delete by update of cancel point logic

SPRESENSE, 2023-11-21:
This pull request https://github.com/apache/nuttx/pull/11165 causes the task to fail to delete.

There is a gnss sensor driver, which creates a task with `open()` and deletes it with `close()`.
[nuttx/boards/arm/cxd56xx/drivers/sensors/cxd5610_gnss.c](https://github.com/apache/nuttx/blob/master/boards/arm/cxd56xx/drivers/sensors/cxd5610_gnss.c)

```
open()
 -> cxd5610_gnss_open()
  -> cxd5610_gnss_initialize()
   -> task_spawn()

close()
 -> cxd5610_gnss_close()
  -> cxd5610_gnss_finalize()
   -> nxtask_delete()
```

In `nxtask_delete()` function, `nxnotify_cancellation()` returns `true` and `nxtask_terminate()` is not executed.
https://github.com/apache/nuttx/blob/11b3adf17fd4e252d880bd9f4dcab54e42e990fe/sched/task/task_delete.c#L123-L134

In `nxnotify_cancellation()` function, the cancellation is pending under the following conditions.
https://github.com/apache/nuttx/blob/11b3adf17fd4e252d880bd9f4dcab54e42e990fe/sched/task/task_cancelpt.c#L106-L127

This was correctly deleting tasks before the pull request https://github.com/apache/nuttx/pull/11165.

I would appreciate any advice.

Thanks.

pussuw, 2023-11-21:
@SPRESENSE sorry about the regression..

Yes, I'll start looking immediately. I can not find an example config with the sensor SENSORS_CXD5610_GNSS enabled, so a  few questions:
- Do you use cancellation points, i.e. CONFIG_CANCELLATION_POINTS=y ?
- Do you define / use up_tls_info() or can you imagine a reason why FAR struct tls_info_s *tls = tls_get_info_pid(tcb->pid); could fail in this case

The only situation where I can imagine that condition fails, is if the tls points to the wrong task OR the tls is corrupt (stack overflow).

pussuw, 2023-11-21:
One last thing that comes to mind, my assumption was that the stack frame created for TLS is memset to 0 always

```
int tls_init_info(FAR struct tcb_s *tcb)
{
  FAR struct tls_info_s *info;

  /* Allocate thread local storage */

  info = up_stack_frame(tcb, tls_info_size());
  if (info == NULL)
    {
      return -ENOMEM;
    }

  DEBUGASSERT(info == tcb->stack_alloc_ptr);

  /* Initialize thread local storage */

  up_tls_initialize(info);

  /* Attach per-task info in group to TLS */

  info->tl_task = tcb->group->tg_info;
  return OK;
}

struct tls_info_s
{
  FAR struct task_info_s * tl_task;

#if defined(CONFIG_TLS_NELEM) && CONFIG_TLS_NELEM > 0
  uintptr_t tl_elem[CONFIG_TLS_NELEM]; /* TLS elements */
#endif

#if defined(CONFIG_PTHREAD_CLEANUP_STACKSIZE) && CONFIG_PTHREAD_CLEANUP_STACKSIZE > 0
  /* tl_tos   - The index to the next available entry at the top of the
   *            stack.
   * tl_stack - The pre-allocated clean-up stack memory.
   */

  uint8_t tl_tos;
  struct pthread_cleanup_s tl_stack[CONFIG_PTHREAD_CLEANUP_STACKSIZE];
#endif

  uint8_t tl_cpstate;                  /* Cancellation state */

#ifdef CONFIG_CANCELLATION_POINTS
  int16_t tl_cpcount;                  /* Nested cancellation point count */
#endif

  int tl_errno;                        /* Per-thread error number */
};
```

However, if this is not the case, and tl_cpstate is not set / memset to 0, you will get undefined behavior.

SPRESENSE, 2023-11-21:
@pussuw Thanks for your investigation.

My conditions are:
* CONFIG_CANCELLATION_POINTS=n
* CONFIG_SCHED_THREAD_LOCAL=n

I found that `nxsched_get_stackinfo()` called from `tls_get_info_pid()` returned `-EACCESS`.
https://github.com/apache/nuttx/blob/eddd90de7865be7bb67217c9d7a3c68c85f1590b/sched/sched/sched_get_stackinfo.c#L86-L98

This may be caused that the task is created by `task_spawn()` with `open()`. 
However, when the task was created by `kthread_create()` instead of `task_spawn()`, it was not possible to delete the task from the application with `close()`.

It used to work this way of implementation, any advice on how to implement it?

pussuw, 2023-11-21:
Hi @SPRESENSE funnily enough, I found this exact line myself just 5 seconds ago :)

I will think of a patch to fix this. The security test (groups are equal) does not make sense in this case, since it is the KERNEL that is asking for the TLS entry, not the user. The kernel should always get access to it.

I had not thought of this security test, I presumed nxsched_get_stackinfo would just give any TLS to me...

pussuw, 2023-11-21:
Once the regression is patched I must think if this "peeking to another process's stack" is a security concern which requires re-designing how nxnotify_cancellation is handled.

But that is irrelevant for fixing the regression and I will fix it later.

pussuw, 2023-11-21:
@SPRESENSE 

This should fix your issue https://github.com/apache/nuttx/pull/11241/

Sorry about the regression, did not test it well enough..

SPRESENSE, 2023-11-22:
@pussuw 
I could confirm that this issue was fixed by https://github.com/apache/nuttx/pull/11241.

SPRESENSE, 2023-11-23:
Close as completed.

closed 2023-11-23 by SPRESENSE