apache/nuttx #11239: Math library header files are not exported

SPRESENSE, 2023-11-21:
When newlib, libmcs, or openlibm is selected as the math library,
the required public header files are not exported by `make export`.
* CONFIG_LIBM_NEWLIB
* CONFIG_LIBM_LIBMCS
* CONFIG_LIBM_OPENLIBM

Refer to the following, these header files should also be exported.
https://github.com/apache/nuttx/blob/11b3adf17fd4e252d880bd9f4dcab54e42e990fe/tools/Config.mk#L660-L675

Alternatively, it is better to link the respective include folders under nuttx/include as well as libcxx.

acassis, 2023-11-22:
@xiaoxiang781216 do you think it was caused by separation of libm to support more libm alternatives?

xiaoxiang781216, 2023-11-23:
yes, the header files provided by new alternatives need copy when "make export".

closed 2023-11-23 by pkarashchenko