apache/nuttx #11267: [bug] cmake building failed

zyy37, 2023-11-25:
[image]
I compile on Linux and arm64 machines, but using cmake will result in compilation failure. The variable `./build/.config:103:CONFIG_HOST_X86_64=y` value in the `.config` file generated by cmake was found to be incorrect and cannot be set by cmake. I'm not sure if there are any other mistakes. The `configure.sh` tool can automatically generate, but it cannot generate files in the `build` folder and cannot be used by cmake.

acassis, 2023-11-27:
@zyy37 thank you for reporting the issue! I think the issue is:
```
choice
        prompt "Host CPU Type"
        default HOST_X86_64
```

So, currently NuttX building system (both Makefile and CMake) are not able to autodetect the host for SIM target.
Maybe if you run "make menuconfig" before "cmake --build build" it will be fixed. Unfortunately I don't have an ARM64 Mac to confirm it.

@pkarashchenko or @cederom could you please confirm?

zyy37, 2023-11-28:
Perhaps Makefile has more comprehensive features, so I will try to use it as much as possible. Thanks

casaroli, 2026-11-08:
Possibly fixed by #13175.