apache/nuttx #11284: [question] Since nuttx and uboot have the same .config configuration and macros, do they also overlap in functionality?

zyy37, 2023-11-28:
I use `nuttx` to build for a certain development board, and use [`uboot`](https://github.com/u-boot/u-boot) (which is a BootLoader tool) to build for the same development board. Both will generate a `.config` configuration file with the same name, which also contains macros with similar names about CPU, SPI, and Device. Since `nuttx` and `uboot` have the same `.config` configuration and macros, do they also overlap in functionality? Could you explain it in detail? Thanks.
for nuttx:
```shell
~/nuttxspace/nuttx$ tools/configure.sh -l stm32f769i-disco:nsh
```
for uboot:
```shell
~/u-boot$ make stm32f769-disco_defconfig
```
[image]

acassis, 2023-11-28:
@zyy37 NuttX uses similar Kconfig frontend like U-Boot and Linux, so they all save a .config. Aside from that they don't overlap in functionality. Although NuttX can do bring-up HW initialization like U-Boot, I think U-Boot normally is used for high-end processors. For low-end MCUs, the comparison could be done with MCUboot. But in this case the MCUboot purpose is mostly to do secure boot and OTA update.

acassis, 2023-12-01:
@zyy37 may I close this question? Since it is not actually an issue, but a curiosity.