apache/nuttx #11304: CMake behavior for `rv-virt:nsh`

yf13, 2023-12-02:
For the  `rv-virt:nsh` target, make approach works well, but CMake build leads to undefined symbols for `g_symtab` etc.  It seems that `defconfig` has `CONFIG_ALLSYMS` off and `CONFIG_NSH_SYMTAB` on. 

Further tests show that:

- When `CONFIG_NSH_SYMTAB` is on and `CONFIG_ALLSYMS` is off, Make build works but CMake fails.
- When both `CONFIG_NSH_SYMTAB` and `CONFIG_ALLSYMS` are off, CMake build works.
- When both  `CONFIG_NSH_SYMTAB` and `CONFIG_ALLSYMS` are on, Makefile build works out a larger binary but CMake build leads to undefined `g_symtab` etc.

Is this issue in current CMake system or something else?

yf13, 2023-12-02:
@raiden00pl thanks a lot. The above point explains CMake's undefined errors  when `CONFIG_ALLSYMS=y` . 

How does  Makefile system works well when `CONFIG_ALLSYMS=n` but `CONFIG_NSH_SYMTAB=y`? Does it contain logic so that undefined errors don't happen?

raiden00pl, 2023-12-02:
It looks like `CONFIG_NSH_SYMTAB` is also not supported for cmake. We have function to generate symbol table in `cmake/nuttx_add_symtab.cmake` but it's not called anywhere and it's quite possible that this logic hasn't been tested by anyone before.

> How does Makefile system works well when CONFIG_ALLSYMS=n but CONFIG_NSH_SYMTAB=y? Does it contain logic so that undefined errors don't happen?

`CONFIG_ALLSYMS` depends on `g_allsyms` and is used for debug purposes (`%pS` and `%ps` format strings for printf, here doc from Linux https://www.kernel.org/doc/html/v4.19/core-api/printk-formats.html#symbols-function-pointers). 

`CONFIG_NSH_SYMTAB` depends on symbol table called `CONFIG_NSH_SYMTAB_ARRAYNAME` which for upstream boards is usually `g_symtab`. 

These are two separate symbol tables, generated in a different way:
- `CONFIG_NSH_SYMTAB` table generated by `apps/tools/mksymtab.sh` (or mentioned earlier `cmake/nuttx_add_symtab.cmake`)
- `CONFIG_ALLSYMS` table generated by `nuttx/tools/mkallsyms.{py|sh}`