apache/nuttx #11314: `ostest`  in `rv-virt:fb` got illegal instruction exception

yf13, 2023-12-04:
When trying the `ostest` within the `rv-virt:fb` configuration, I got illegal instructions exceptions like below:

```
user_main: FPU test
Starting task FPU#1
fpu_test: Started task FPU#1 at PID=6
[    9.170000] riscv_exception: EXCEPTION: Illegal instruction. MCAUSE: 00000002, EPC: 80022086, MTVAL: 00000000
[    9.170000] riscv_exception: PANIC!!! Exception = 00000002
[    9.170000] _assert: Current Version: NuttX  0.0.0 95ac627d59-dirty Dec  4 2023 09:41:38 risc-v
[    9.170000] _assert: Assertion failed panic: at file: :0 task: FPU#1 process: FPU#1 0x80021ff4
[    9.170000] up_dump_register: EPC: 80022086
[    9.170000] up_dump_register: A0: 00000000 A1: 80179f18 A2: 80179f18 A3: 00000002
[    9.170000] up_dump_register: A4: 8003ce8c A5: 80032000 A6: 00000000 A7: 00000000
[    9.170000] up_dump_register: T0: 00000000 T1: 00000000 T2: 00000000 T3: 00000000
[    9.170000] up_dump_register: T4: 00000000 T5: 00000000 T6: 00000000
[    9.170000] up_dump_register: S0: 00000000 S1: 00000001 S2: 00000000 S3: 00000000
[    9.170000] up_dump_register: S4: 00000000 S5: 8003ce8c S6: 00000000 S7: 00000000
[    9.170000] up_dump_register: S8: 00000000 S9: 00000000 S10: 00000000 S11: 00000000
[    9.170000] up_dump_register: SP: 8017a660 FP: 00000000 TP: 00000000 RA: 8002203a
[    9.170000] dump_stack: User Stack:
[    9.170000] dump_stack:   base: 0x80179f28
[    9.170000] dump_stack:   size: 00002008
[    9.170000] dump_stack:     sp: 0x8017a660
[    9.170000] stack_dump: 0x8017a660: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    9.170000] stack_dump: 0x8017a680: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    9.170000] stack_dump: 0x8017a6a0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[    9.170000] stack_dump: 0x8017a6c0: 00000000 00000000 80021ff4 800061cc 00000000 00000000 80179f18 00000001
[    9.170000] stack_dump: 0x8017a6e0: 00000000 00000000 00000000 80001962 00000000 00000000 00000000 00000000
[    9.170000] dump_tasks:    PID GROUP PRI POLICY   TYPE    NPX STATE   EVENT      SIGMASK          STACKBASE  STACKSIZE      USED   FILLED    COMMAND
[    9.170000] dump_tasks:   ----   --- --- -------- ------- --- ------- ---------- ---------------- 0x800321a0      2048       716    34.9%    irq
[    9.170000] dump_task:       0     0   0 FIFO     Kthread - Ready              0000000000000000 0x8003d68c      2032       644    31.6%    Idle_Task
[    9.170000] dump_task:       1     1 224 RR       Kthread - Waiting Semaphore  0000000000000000 0x8003f0b0      1984       540    27.2%    hpwork 0x80032e04 0x80032e18
[    9.170000] dump_task:       2     2 100 RR       Kthread - Waiting Semaphore  0000000000000000 0x8003fcb8      1976       540    27.3%    lpwork 0x80032ddc 0x80032df0
[    9.170000] dump_task:       3     3 100 RR       Task    - Waiting Signal     0000000000000000 0x800408e8      3016      1404    46.5%    nsh_main
[    9.170000] dump_task:       4     4 100 RR       Task    - Waiting Signal     0000000000000000 0x80177038      2008       956    47.6%    ostest
[    9.170000] dump_task:       5     5 100 RR       Task    - Waiting Signal     0000000000000000 0x80177b78      8104       764     9.4%    ostest Arg1 Arg2 Arg3 Arg4
[    9.170000] dump_task:       6     6 100 RR       Task    - Running            0000000000000000 0x80179f28      2008       556    27.6%    FPU#1
```

masayuki2009, 2023-12-05:
> @masayuki2009 any idea about this issue?

@yf13 
I have no idea but please provide the following information when you report a bug.

Host OS name and CPU type.
Toolchain version.
QEMU version and runtime options.
NuttX code base  (branch name and the commit id)

yf13, 2023-12-05:
@masayuki2009, thanks for your attention.

Host info:  Ubuntu 22.04 Jammy on AMD64 PC
Toolchain info: xpack-riscv-none-elf-gcc-13.2.0-2-linux-x64.tar.gz
QEMU info: qemu-system-riscv32 from zephyr-sdk-0.16.4. It shows version 7.0.0 (v7.0.0-6-g2cc2e86de6-dirty). 
QEMU command line to run NuttX:

```
$ qemu-system-riscv32 -semihosting -M virt -smp 2 -chardev stdio,id=con,mux=on -serial chardev:con -device virtio-gpu-device,xres=640,yres=480,bus=virtio-mmio-bus.0 -mon chardev=con,mode=readline -device loader,file=nuttx
VNC server running on 127.0.0.1:5900

NuttShell (NSH) NuttX-12.3.0
nsh> uname -a
NuttX 12.3.0 95ac627d59-dirty Dec  5 2023 14:31:50 risc-v rv-virt

```
Nuttx codebase: master branch from upstream with commit id 95ac627d59

masayuki2009, 2023-12-05:
@yf13 

Thanks for the information.
Though I tried the same settings, I was not able to reproduce the crash.

As for the nuttx code base, you wrote

>Nuttx codebase: master branch from upstream with commit id https://github.com/apache/nuttx/commit/95ac627d590e5825ac9f8c969e87694ad5e5afd3

However, uname -a shows 95ac627d59-dirty which means you modified the source code.
Please confirm both nuttx and apps directory and where you modified the source code.

yf13, 2023-12-06:
@masayuki2009, thanks for the information.

Here my modification are for the download URLs in `openamp/open-amp.defs` and `openamp/libmetal.defs` to avoid unstable Github connections here.

masayuki2009, 2023-12-06:
@acassis 
As I commented above, I was not able to reproduce the issue.
Could you try the latest nuttx with your PC to check if the issue is happening?

yf13, 2024-05-02:
Checked with recent build and it passed:

```
nsh> cat /proc/version
NuttX version 12.4.0 639c053289 May  3 2024 07:26:03 rv-virt/fb-dirty
nsk> free
                 total       used       free    maxused    maxfree  nused  nfree
      Umem:   33298060    1282804   32015256    2586604   31994992     55      6
nsh> ostest
...
Final memory usage:
VARIABLE  BEFORE   AFTER
======== ======== ========
arena     1fc168c  1fc168c
ordblks         2        4
mxordblk  1e85d18  1e83470
uordblks   13b8fc   13c9cc
fordblks  1e85d90  1e84cc0
user_main: Exiting
ostest_main: Exiting with status 0
```

We can close this for now.

closed 2024-05-02 by yf13