apache/nuttx #11321: Simplify NuttX initialization

acassis, 2023-12-04:
Currently NuttX has many initialization options each one used for some specific case, but at least CONFIG_NSH_ARCHINIT that is used to do drivers initialization when NSH is enabled could be removed in favor of some generic option like BOARD_LATE_INITIALIZATION that will work for all applications (inclusive to NSH itself).

I raised this question some time ago here: https://github.com/apache/nuttx/issues/10821 but didn't heard anything from my peers.

But now that @0528Mike came with this PR: https://github.com/apache/nuttx/pull/11306 we could fix this issue merging CONFIG_NSH_ARCHINIT and BOARD_LATE_INITIALIZATION to avoid issues like this faced from a NuttX user:  https://github.com/apache/nuttx/issues/10821 and for many people that want to user their application as ENTRYPOINT on NuttX.

xiaoxiang781216, 2023-12-05:
Yes, I think the suggestion is reasonable.

raiden00pl, 2023-12-07:
+1 from me

raiden00pl, 2025-01-06:
hi, are we trying to solve this problem? Does anyone have any objections to removing `CONFIG_NSH_ARCHINIT` ?
This option is slowly starting to cause problems in the case of PCI devices. Some OS components that have a PCI based virtio backend need to be initialized in the correct order (e.g. coredump with PCI based virtblk device), which is impossible to achieve with `CONFIG_NSH_ARCHINIT`.

xiaoxiang781216, 2025-01-06:
NSH_ARCHINIT can be achieved in board initializiton code, so it could be removed to avoid the confusion.

gregory-nutt, 2025-01-06:
> NSH_ARCHINIT can be achieved in board initializiton code, so it could be removed to avoid the confusion.

Agreed.  But removing CONFIG_NSH_ARCHINIT will be a very big job because it is used by 1246 board configurations.

```
$ find boards -name defconfig | xargs grep CONFIG_NSH_ARCHINIT=y | wc -
   1246    1246  100027 
```
That would be a LOT of breakage!

raiden00pl, 2025-01-18:
Yes, it seems like a lot of work. But it will only get worse over time as new boards come in.

We can automate most of the work. What we need to do is:

1. remove all `CONFIG_NSH_ARCHINIT` from configs 
2. add `CONFIG_BOARD_LATE_INITIALIZATION` to all modified configs
3. clean up after removing `CONFIG_NSH_ARCHINIT`.

Is there anything more to do ?

The last step most likely must be done manually. But there are "only" 115 files that need modification:

```
$ find . -type f ! -name "defconfig" -exec grep -l "CONFIG_NSH_ARCHINIT" {} + | wc                                                                                                                              
    115     115    6899
```

gregory-nutt, 2025-01-18:
> Is there anything more to do ?

I think a proper solution would be the complete remove of the archi-init logic, not just disabling it.  I think that would cause a lot of confusion in the future.

This would require removal the the BOARD_CTRL log, removal of the arch-init entry point to the board initialization logic, and removal of all calls to the removed BOARD_CTRL logic.

raiden00pl, 2025-01-22:
It makes sense. Such a change deserve major version increase during release, as this is substantial change for users. Probably this issue is a good milestone for NuttX 13.0.0 release

linguini1, 2026-02-17:
Some boards don't appear to have a `board_late_initialize` defined, only the app_init logic. For those my current plan is to take the contents of the `board_app_initialize` function and put it in a new `board_late_initialize` function.

lupyuen, 2026-02-19:
FYI I have a couple of articles explaining the NuttX Boot Flow...

- [__NuttX Boot Flow in PDF__](https://github.com/lupyuen/nuttx-arm64-emulator/blob/avaota/nuttx-boot-flow.pdf) / [__SVG__](https://github.com/lupyuen/nuttx-arm64-emulator/blob/avaota/nuttx-boot-flow.svg) / [__PNG__](https://github.com/lupyuen/nuttx-arm64-emulator/blob/avaota/nuttx-boot-flow.png)

- Based on the [__Mermaid Flowchat__](https://raw.githubusercontent.com/lupyuen/nuttx-arm64-emulator/refs/heads/avaota/nuttx-boot-flow.mmd)

- Which was generated by our [Arm64 Avaota-A1 Emulator](https://lupyuen.org/articles/unicorn4)

We also fed the above [__Mermaid Flowchat__](https://raw.githubusercontent.com/lupyuen/nuttx-arm64-emulator/refs/heads/avaota/nuttx-boot-flow.mmd) to Gemini LLM to get these...

- [__LLM explains NuttX Boot Flow (Short Version)__](https://gist.github.com/lupyuen/b7d937c302d1926f62cea3411ca0b3c6)

- [__LLM explains NuttX Boot Flow (Long Version)__](https://docs.google.com/document/d/1qYkBu3ca3o5BXdwtUpe0EirMv9PpMOdmf7QBnqGFJkA/edit?tab=t.0)

See also:

- [NuttX Boot Flow for RISC-V Ox64 BL808](https://lupyuen.org/articles/ox2#appendix-nuttx-boot-flow)

- [NuttX Boot Flow for Arm64 PinePhone](https://lupyuen.org/articles/unicorn2#pinephone-boots-nuttx)

hartmannathan, 2026-02-19:
> @linguini1 is the boot process is a mess, there are originally the xxx_boot.c and later xxx_bringup.c. Later xxx_appinit.c and finally there is only one xxx_init.c: avr/avrdx/breadxavr/src/avrdx_init.c
> 
> I think we need some consensus before changing it! For visibility: @raiden00pl @simbit18 @hartmannathan @xiaoxiang781216 @lupyuen  @jerpelea @cederom (sorry if I missed another people interested on it)

Hi all,

I'm in favor of improving the boot process. Just please remember that 3rd party developers may have private, out-of-tree boards that utilize some of these functions in their initialization. (My boards use board_late_initialize, but others may have done differently.) So whatever approach we take to simplify and clean up the boot process, we need to make sure that developers can have the hooks they need to initialize their stuff properly. And we need to make sure that any changes people will need to make, will be properly documented in the release notes!!!