apache/nuttx #11337: esp32s3-devkit: spi flash initialize failed

dapeng0512, 2023-12-07:
Hi, 

There may be a bug. the spi flash initialize failed

This is the bringup log:

```
-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x1664
load:0x403c9700,len:0x4
load:0x403c9704,len:0xb74
load:0x403cc700,len:0x2e8c
entry 0x403c98fc
I (27) boot: ESP-IDF v5.1-dev-3972-g1559b6309f 2nd stage bootloader
I (27) boot: compile time Mar 15 2023 12:14:10
I (28) boot: chip revision: v0.1
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 factory          factory app      00 00 00010000 00100000
I (69) boot: End of partition table
I (74) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=029c8h ( 10696) map
I (84) esp_image: segment 1: paddr=000129f0 vaddr=3fc896f0 size=00a20h (  2592) load
I (91) esp_image: segment 2: paddr=00013418 vaddr=40374000 size=02cb0h ( 11440) load
I (101) esp_image: segment 3: paddr=000160d0 vaddr=00000000 size=09f48h ( 40776) 
I (114) esp_image: segment 4: paddr=00020020 vaddr=42000020 size=162b0h ( 90800) map
I (133) boot: Loaded app from partition at offset 0x10000
I (133) boot: Disabling RNG early entropy source...
ABERROR: Failed to initialize SPI Flash

NuttShell (NSH) NuttX-12.3.0
nsh> 
```

I am using the official examples test, the specific steps are as follows：

1. tools/configure.sh esp32s3-devkit:spiflash
2. make
3. make flash

BTY: I remember that when I use mcuboot, I need to open XMC support, I don't know if this is also the problem

Thanks.

acassis, 2023-12-08:
@dapeng0512 did you try some older version? Maybe the first version with ESP32-S3 SPI Flash support.

@tmedicci did you see it happening too?

tmedicci, 2023-12-10:
Hi @dapeng0512 , we need more information.

Which board are you using? (especially, which module). Also, can you please run the [`nxdiag`](https://nuttx.apache.org/docs/latest/applications/system/nxdiag/index.html) application with [Espressif's specific configs](https://nuttx.apache.org/docs/latest/applications/system/nxdiag/config.html#nxdiag-specific-configuration-settings) enabled too.

dapeng0512, 2023-12-11:
Hi @tmedicci ,  I test on ESP32-S3 devkit with `ESP32-S3-WROOM-1 M0N8R2` module. this is all serial log and `nxdiag --all` result

[log_20231211_1653.log](https://github.com/apache/nuttx/files/13633089/log_20231211_1653.log)

acassis, 2023-12-11:
@tmedicci maybe using littlefs could avoid this issue, because AFAIK lfs could work even when flash is not formatted.

tmedicci, 2023-12-11:
> @tmedicci maybe using littlefs could avoid this issue, because AFAIK lfs could work even when flash is not formatted.

We usually use the SPIFFS for other purposes, but our internal CI tests the SmartFS, that's why it's the default ;)

dapeng0512, 2023-12-13:
All right, thanks for your reply @tmedicci. manual formatting is indeed required.

closed 2023-12-18 by xiaoxiang781216