apache/nuttx #11368: [help] how to set CONFIG_DISABLE_IDLE_LOOP in defconfig

yf13, 2023-12-11:
I have  `CONFIG_DISABLE_IDLE_LOOP=y` added to my defconfig, but after running `tools/configure.sh`, the option is not in resulted `.config`. Thus I have to use `kconfig-tweak --enable DISABLE_IDLE_LOOP` to set it each time. 

How to correctly set  it in my defconfig?

xiaoxiang781216, 2023-12-11:
You can't enable DISABLE_IDLE_LOOP from defconfig, since DISABLE_IDLE_LOOP is designed to be selected by other options in Kconfig, for examples SYSTEM_OFLOADER:
https://github.com/apache/nuttx-apps/blob/master/system/ofloader/Kconfig#L9

xiaoxiang781216, 2023-12-11:
it's inside Linux kernel: https://www.kernel.org/doc/html/latest/kbuild/kconfig-language.html

yf13, 2023-12-11:
I see. I was wondering there might be some differences.

closed 2023-12-11 by yf13

xiaoxiang781216, 2023-12-11:
No, NuttX' config is same as Linux.

yf13, 2023-12-12:
I guess this [copy](https://bitbucket.org/nuttx/tools/src/master/kconfig-language.txt) it is. :)