apache/nuttx #11376: cxxfilt was included in official NuttX documentation install but is not present on Ubuntu

acassis, 2023-12-12:
I receive a comment from a user trying to install NuttX dependencies and cannot figure out how to install cxxfilt because pip message recommends to run apt install python3-cxxfilt but this packet doesn't exist on Ubuntu.

More info:
https://www.embeddedrelated.com/showarticle/1524.php#comments

@xiaoxiang781216 any idea how to solve this issue?

xiaoxiang781216, 2023-12-13:
I am using Ubuntu 22.04, the installation run without problem. @Gary-Hobson do you encounter the similar problem?

Gary-Hobson, 2023-12-13:
I am using ubuntu 22.04, but I reproduced this problem in docker

I found the reason and solution on stackoverflow: https://stackoverflow.com/questions/75602063/pip-install-r-requirements-txt-is-failing-this-environment-is-externally-mana

The recommended solution is:
```
python3 -m venv .venv
source .venv/bin/activate
pip install cxxfilt
```
A simpler way is:
```
pip install --break-system-packages --user cxxfilt
```

acassis, 2023-12-13:
Thank you @Gary-Hobson and @xiaoxiang781216

closed 2023-12-13 by acassis