apache/nuttx #11390: Add support to use NuttX as POSIX PSE51

acassis, 2023-12-13:
NuttX is a very complete POSIX RTOS, but sometimes for low end applications like some low cost automotive application POSIX standard is overwhelming.

This issue was fixed with PSE51 subset, the doesn't require many POSIX features, like file systems, etc.

Some years ago I removed many of NuttX features to make it running in a low end DIP package LPC1114 MCU, picture here:
https://flickr.com/photos/79703543@N00/18013069041/in/dateposted-public/

After some time this minnsh demo was removed from NuttX mainline and many options to disable feature also were removed from kernel (option to disable FILES, VFS, etc). It is important to remove this option to disable NuttX features to make it adaptable to POSIX PSE51 standard.

xiaoxiang781216, 2023-12-14:
it's a very interesting topic, we are looking to fully support PSE51 spec.

raiden00pl, 2023-12-14:
The ability to run NuttX on low cost chips would be a very cool feature, but if we're thinking about PSE51, then supporting the rest of POSIX RT profiles would be a good idea.
If we could select supported POSIX features by just setting the appropriate configuration option, it would be perfect.
It's very interesting how low it's possible to go for NuttX with the required chip resources.

acassis, 2024-07-13:
The POSIX missing functions can be found here:
https://medium.com/@zim.kalinowski/posix-support-comparison-in-various-rtoses-b45677889ec8

acassis, 2024-11-22:
@xiaoxiang781216 @raiden00pl I think we need to rollback that patches that disabled the ability to disable signals, to remove VFS support, etc. The PSE51 doesn't require it. Please see the discussion at Discord: https://discord.com/channels/716091708336504884/716091708819111959

raiden00pl, 2024-11-23:
I'll deal with this issue sometime at the beginning of next year. Is there any discussion or justification for removing these options (DISABLE_SIGNALS etc) in the past ? I think I saw something like this before, but I can't remember where.

raiden00pl, 2025-02-05:
it's useful to disable signals, but it's against POSIX, event PSE51 requires signals. Maybe if we could help the compiler optimize unused signal logic, but I don't know if that's possible.

xiaoxiang781216, 2025-02-05:
But I think it's different thing, the user could unselect DISABLE_SIGNALS to keep the compliant at any time.
We even have option to disable timer/pthread/mqueu(https://github.com/apache/nuttx/blob/master/sched/Kconfig#L6-L47), so it is reasonable to restore disable signal option here.

acassis, 2025-02-05:
I think it should be nice to have support to original "minnsh" where almost everything were disabled, even signals, VFS, etc