apache/nuttx #11420: use SDCard on NuttX question。

GC-20-20, 2023-12-19:
I tried to follow the tutorial to use the SD card on the nuttx, but now it reports this error when mounting the SD:
[image]
I can't find what type of error this is.

acassis, 2023-12-20:
@GC-20-20 Please share more information about your setup. What board are you using, what Ubuntu version, what toolchain, etc.

This number 5 means I/O Error, please look at nuttx/include/errno.h

GC-20-20, 2023-12-21:
> @GC-20-20 Please share more information about your setup. What board are you using, what Ubuntu version, what toolchain, etc.请分享有关您的设置的更多信息。你用的是什么板子，什么Ubuntu版本，什么工具链，等等。
> 
> This number 5 means I/O Error, please look at nuttx/include/errno.h这个数字 5 表示 I/O 错误，请查看 nuttx/include/errno.h

> @GC-20-20 Please share more information about your setup. What board are you using, what Ubuntu version, what toolchain, etc.请分享有关您的设置的更多信息。你用的是什么板子，什么Ubuntu版本，什么工具链，等等。
> 
> This number 5 means I/O Error, please look at nuttx/include/errno.h这个数字 5 表示 I/O 错误，请查看 nuttx/include/errno.h

Thank you very much for your reply, I am using a Chinese learning development board with STM32F429, I rechecked the SDIO pins and the configuration information, but it still can't be mounted with ERROR: 5. Do you have any idea in this case?
[image]

acassis, 2023-12-22:
@GC-20-20 please double check you board config, I suggest you to look the SDCARD examples for stm32f4discovery and stm32f401rc-rs485. Double checkout all pins and notice that your board doesn't have the Card Dection (CD) pin connected to the MCU. 

Also you can enable the debug options to collect more information:

```
Build Setup  --->
    Debug Options  --->
        [*] Enable Debug Features
        ...
        [*]   File System Debug Features
        [*]     File System Error Output
        [*]     File System Warnings Output
        ...
        [*]   Memory Card Driver Debug Features
        [*]     Memory Card Driver Error Output
        [*]     Memory Card Driver Warnings Output
        [*]     Memory Card Driver Informational Output
```

GC-20-20, 2023-12-23:
@acassis Thanks for your reply, I have now successfully mounted it. The cause was a hardware problem, two modules were mounted on the SDIO bus, masking one of them worked.

closed 2023-12-23 by xiaoxiang781216