apache/nuttx #11424: Bug when running `make olddefconfig V=1`

tmedicci, 2023-12-19:
After https://github.com/apache/nuttx/pull/10801, running the command `make olddefconfig` with `V=1` verbosity fails in some environments. Specifically, it fails with the NuttX CI container (https://github.com/apache/nuttx/pkgs/container/nuttx%2Fapache-nuttx-ci-linux) and, possibly, on Ubuntu 22.04 (as it's the container`s base image) with the following error message:

```
APPSDIR="../apps" EXTERNALDIR=dummy APPSBINDIR="../apps" BINDIR=/nuttx olddefconfig 2> >(grep -v "warning: the 'modules' option is not supported" | tee kwarning) | cat && if [ -s kwarning ]; then rm kwarning; exit 1; else rm kwarning; fi
/bin/sh: 1: Syntax error: redirection unexpected
make: *** [tools/Unix.mk:688: olddefconfig] Error 2
```

Running it locally on Manjaro, it works as expected.

acassis, 2023-12-20:
@xiaoxiang781216 @pkarashchenko @raiden00pl any idea why is it happening?

xiaoxiang781216, 2023-12-20:
@xuxin930 could you take a look?

xuxin930, 2023-12-21:
> @xuxin930 could you take a look?

sure

xuxin930, 2023-12-21:
this issue is caused by the verbosity option `V` restricting the shell type.
https://github.com/apache/nuttx/blob/a506f9fb709fa7d3d6646d6c9ffd2246ded8e377/tools/Config.mk#L41-L44

I adjusted the judgment logic in this PR , please review https://github.com/apache/nuttx/pull/11432

closed 2023-12-21 by xiaoxiang781216