apache/nuttx #11435: Make OPEN option select FIFO and named pipes configuration

gus-soares19, 2023-12-21:
When enabling open()/pclose() functions, in order to use them, 'FIFO and named pipes drivers' option must be enabled but there is no sign of its dependecy. Making 'popen()/pclose() Functions' option select 'FIFO and named pipes drivers' when enabled would solve it.

xiaoxiang781216, 2023-12-21:
it's better to use `depends on` not `select`.

hartmannathan, 2023-12-21:
This issue was noticed during this email thread: https://lists.apache.org/thread/fnpv47sslyyzzjwkw6ydmdwclf8vfqzj

acassis, 2023-12-21:
@gus-soares19 thank you for reporting about it at NuttX mailing list and for opening the issue here.

> it's better to use `depends on` not `select`.

Currently it is 'depends on' it should be 'selects', I'm going to send a patch

gus-soares19, 2023-12-22:
No, I did not. I enabled both CONFIG_EXAMPLES_POPEN and CONFIG_SYSTEM_POPEN using menuconfig.

xiaoxiang781216, 2023-12-23:
But you must enable CONFIG_PIPES in defconfig, otherwise CONFIG_EXAMPLES_POPEN/CONFIG_SYSTEM_POPEN will be ignored by kconfig tool since the dependence requirement isn't matched. It's strange that you hit a linker error, not lose popen function and example. Could you reproduce the problem in a clear repo?

gus-soares19, 2023-12-23:
> Could you reproduce the problem in a clear repo?

Using a new installation, trying to search for _CONFIG_EXAMPLES_POPEN_ and _CONFIG_SYSTEM_POPEN_, I couldn't enable them.

this is SYSTEM_POPEN Kconfig in the new instalation:
[image]

and here in my project configuration:
[image]

there is no declared dependency on _PIPES_.

acassis, 2023-12-23:
@gus-soares19 you cannot find them because of the "depends on". When a feature A "depends on" feature B, it will not be displayed until you enable first the feature B.

So in this case you need to enable PIPES first, then SYSTEM_POPEN and only later EXAMPLES_POPEN.

And that is the point! Since you never enabled CONFIG_PIPES in your previous configuration the only way you could have enabled SYSTEM_POPEN and EXAMPLES_POPEN was if you edited the .config file manually.

So, you broke the failsafe protection of menuconfig ! :-)

gus-soares19, 2023-12-23:
But as shown in the second image in my previous reply, there is no dependency on _PIPES_, only requiring _NSH_LIBRARY_ to be enabled. This way I was capable of enabling _SYSTEM_POPEN_ without _PIPES_, causing the error I was facing. In the new installation it is fixed somehow, so the problem is solved I guess.

acassis, 2023-12-23:
@xiaoxiang781216  currently we have the "-dirty" feature that @lucasssvaz implemented to let us know when the user edited the default config using make menuconfig. Maybe we need to have some similar feature to let us to know when the user modified the .config manually instead using menuconfig. What do you think? Any idea how we could do it properly?

gus-soares19, 2023-12-23:
It would help to find mistakes made by users, great idea, but no idea how to implement it.

lucasssvaz, 2023-12-23:
One (naive) way of doing that is making the menuconfig save the timestamp of when it was last edited and then comparing with the actual timestamp of the .config file. Idk if this would be currently be doable without editing the menuconfig source (I guess it could be done in a similar way that the "-dirty" tag is added through the configure script).

acassis, 2023-12-24:
> But as shown in the second image in my previous reply, there is no dependency on _PIPES_, only requiring _NSH_LIBRARY_ to be enabled. This way I was capable of enabling _SYSTEM_POPEN_ without [...]

You are right, older version of apps/system/popen/Kconfig didn't have this lock!

acassis, 2023-12-24:
> One (naive) way of doing that is making the menuconfig save the timestamp of when it was last edited and then comparing with the actual timestamp of the .config file. Idk if this would be currently [...]

Yes, that idea could work correctly! Thank you Lucas!!!

I was thinking something like a text checksum at end of the .config file, but maybe the date could be enough!

What do you think @xiaoxiang781216 ?

gus-soares19, 2023-12-24:
> You are right, older version of apps/system/popen/Kconfig didn't have this lock!

That's what I thougth, my project is using a older version of NuttX which don't validate _PIPES_... the current version blocks _SYSTEM_POPEN_ if _PIPES_ is not enabled.

casaroli, 2026-11-08:
Possibly fixed by #20778.

closed 2026-11-08 by casaroli