apache/nuttx #11438: CMake version.h

maxikrie, 2023-12-21:
When using the CMake build system, there seems to be an issue with the generation of version.h. Specifically, I get the following error

`nuttx/build/include/nuttx/version.h:7:1: error: missing terminating " character    7 | "`

caused by the closing apostrophe being wrongly in the next line:

`#define CONFIG_VERSION_STRING "12.3.0`
`"`

After moving the apostrophe to the right line, it compiles. When using the Makefile build system, this issue doesn't occur.

My system is the following:
System: Debian 11 6.1.0-0.deb11.13-amd64
Toolchain: arm-gnu-toolchain-13.2.rel1-x86_64-arm-none-eab
Board:  nucleo-h743zi2
Config: nsh

acassis, 2023-12-22:
@maxikrie thank you for reporting, I think the error is here:
```
cmake/nuttx_mkversion.cmake:
file(APPEND ${VERSION_H} "#define CONFIG_VERSION_STRING \"${NUTTX_VERSION}\"\n")
```

Could you please replace it with:
```
file(APPEND ${VERSION_H} "#define CONFIG_VERSION_STRING \"${NUTTX_VERSION}\"\")
```

And test again.

maxikrie, 2023-12-28:
Thanks for pointing me to the relevant file. Unfortunately, your proposed fix doesn't resolve the issue, but results in 
`#define CONFIG_VERSION_STRING "12.3.0`
`"#define CONFIG_VERSION_MAJOR 12`

It seems like the command in line 26 outputs NUTTX_VERSION with an added "\n" character in the end, which causes the observed issue. I could resolve this by adding the following in line 33
`string(REPLACE "\n" "" NUTTX_VERSION ${NUTTX_VERSION})`

Maybe somebody can suggest a cleaner solution?

In any case, it is great to now have the option of the CMake build!

maxikrie, 2024-01-03:
@acassis any thoughts to my proposal?

pietroalbini, 2024-08-21:
I did hit this while building NuttX from source, opened https://github.com/apache/nuttx/issues/11438 with the fix.

acassis, 2024-08-21:
Thank you @pietroalbini !!! @maxikrie could you please take a look?

closed 2024-08-21 by xiaoxiang781216

maxikrie, 2024-10-03:
The mentioned commit solves the problem, thanks!