apache/nuttx #11442: Failed to Build with Custom Apps Directory

j-couch, 2023-12-23:
I am trying to build with a custom apps directory following this guide:
https://nuttx.apache.org/docs/latest/guides/customapps.html#custom-apps-how-to

Using NuttX Release 12.3.0 in WSL (Linux version 5.15.133.1-microsoft-standard-WSL2), I created the example files listed in the guide and try to build

`> ./tools/configure.sh -E -a ../CustomApps sim:nsh` runs fine
`> make` has a build error:
```
/usr/bin/ld: nuttx.rel: warning: relocation against `g_builtins' in read-only section `.text.builtin_for_index'
/usr/bin/ld: nuttx.rel: in function `builtin_for_index':
/nuttx/libs/libc/builtin/lib_builtin_forindex.c:54: undefined reference to `g_builtin_count'
/usr/bin/ld: /nuttx/libs/libc/builtin/lib_builtin_forindex.c:56: undefined reference to `g_builtins'
/usr/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
```
Looking in normal Apps directory `g_builtin_count` and `g_builtins` are defined in `builtin/builtin_list.c`, does this or something similar need to be added to custom apps directory as well? Or maybe I have a configuration issue unnecessarily requiring `g_builtin_count` and `g_builtins`?

Also just tested with latest `master` @ 804f713 and see the same issue

j-couch, 2023-12-27:
@acassis thank you for the quick response! I was originally trying to recreate the "1. Replace The Apps/ Directory Completely" method but was able to duplicate your steps here, compile and run, and believe the symbolic link method should work okay for my purposes instead. Maybe the "Replace" section should be removed from the guide?

TimJTi, 2023-12-27:
The # issue not copying is some peculiarity of the way the pages are rendered. They copy fine when editing the files and previewing and there is no way of "escaping" that I know of. They are in code blocks. Perhaps someone who knows rst and or the way the NuttX web server renders it can advise.

Not sure why I used "custom_hello rather than "main" - I know I tested it all so either something in NuttX has changed, or I edited the names in the text prior to submission and didn't check again.

Solution 1) should probably be zapped - not having access to the existing nuttx-apps would, I think, leave NuttX severly lacking in features.

Perhaps @j-couch or @acassis could do the PR for these changes.