apache/nuttx #11444: rp2040: mv command hangs up at SMARTFS on internal flash memory

takumiando, 2023-12-25:
## Description

```mv``` command hangs up at SMARTFS on internal flash memory.

## Version

- ```nuttx-12.3.0```

## Patch

I applied this patch to use ```usbnsh``` config with flash file system.

``` patch
diff --git a/boards/arm/rp2040/raspberrypi-pico-w/configs/usbnsh/defconfig b/boards/arm/rp2040/raspberrypi-pico-w/configs/usbnsh/defconfig
index f49cf6b8ba..fdda8f63d1 100644
--- a/boards/arm/rp2040/raspberrypi-pico-w/configs/usbnsh/defconfig
+++ b/boards/arm/rp2040/raspberrypi-pico-w/configs/usbnsh/defconfig
@@ -49,3 +49,5 @@ CONFIG_TESTING_GETPRIME=y
 CONFIG_TESTING_OSTEST=y
 CONFIG_USBDEV=y
 CONFIG_USBDEV_BUSPOWERED=y
+CONFIG_RP2040_FLASH_FILE_SYSTEM=y
+CONFIG_SMARTFS_ALIGNED_ACCESS=y
```

## Build

``` shell
$ ./tools/configure.sh raspberrypi-pico-w/usbnsh
$ make
```

## Run ```mv```

``` shell
NuttShell (NSH) NuttX-12.3.0
nsh> cd /flash
nsh> ls
 test
nsh> mv test test2
*** Hangs up here ***
```

I found that it hangs up at ```smartfs_rename()```, but I couldn't research more than that...
The functions that uses ```smartfs_rename()``` may have same issue (only at this chip?).

acassis, 2023-12-25:
@takumiando Please submit a PR with this fix

takumiando, 2023-12-25:
@acassis Sorry, I haven't found a cause of this problem yet.
This is just a issue report.

acassis, 2023-12-25:
@takumiando can you try to duplicate it using for example stm32f4discovery board? Maybe it could be some missing configuration for rp2040 arch.

takumiando, 2023-12-26:
@acassis I resolved with [this PR](https://github.com/apache/nuttx/pull/11451)!

closed 2023-12-26 by xiaoxiang781216