apache/nuttx #11450: Missing arch/ and configs/ subdirectories when "git checkout" 7.11 to 7.15

acassis, 2023-12-25:
I'm trying to get Atmega128 work with NuttX and when I did "git checkout 7.15" noticed that arch/ and configs/ directory were empty.

I think it was caused because in the past some directories were separated from nuttx/ kernel.

This issue happens from 7.11 to 7.15! So version <= 7.10 seems fine as well version >= 7.16.

@patacongo I remember you said you have all NuttX tarball versions saved in your computer.
Could you please update it to some free online file sharing? (i.e. https://www.pcloud.com offer 10GB for free)

I'm using the BK-AVR128 board and noticed that older NuttX version (i.e. 6.10 before Kconfig era) didn't have any activity in the serial (the TXD line just went from low to high) and the same thing happens after version 7.18.

I'm using avr-gcc toolchain (called Linux GCC on menuconfig), but sure if avr buildroot toolchain works, since I don't have it.

acassis, 2023-12-25:
This is how I discovered to issue:

I'm using the BK-AVR128 board and noticed that older NuttX version (i.e. 6.10 before Kconfig era) didn't have any activity in the serial (the TXD line just went from low to high) and the same thing happens after version 7.17.

In version 7.16 I can see it printing many "KKKKKKKKKKKKKKKK", using logic analyzer the sequence seems to be 0x44 0x4b 0x4b 0x4b 0x4b ....

I'm using avr-gcc toolchain (called Linux GCC on menuconfig), but sure if avr buildroot toolchain works, since I don't have it.

[image]

acassis, 2023-12-25:
More some info:

eac916c90797d1c342e9e9663084b3155d49a481 - doesn't work

ed1535f18846ebcd1744121b85de40dde5d25e3d - prints: DKKKKKKK... in the serial console

patacongo, 2023-12-25:
> @patacongo I remember you said you have all NuttX tarball versions saved in your computer.
> Could you please update it to some free online file sharing? (i.e. https://www.pcloud.com offer 10GB for free)

7.12 through 7.15 are available here: https://sourceforge.net/projects/nuttx/files/nuttx/  The arch/ and configs/ directories are fully present in that tarball.  Sounds like the upstream repository is bad.

If you were looking at a version where the submodules were used, then you would see submdule files in those directory.  They are not so I assume that there is some corruption in the githut repositories.

patacongo, 2023-12-25:
> If you were looking at a version where the submodules were used, then you would see submdule files in those directory. They are not so I assume that there is some corruption in the githut repositories.

The Documentation/ directory was made a submodule by 40f52f72d793ccdfb16cb8076d2d03aeb60c61ca
The configs/ directory was made a submodule by 2113f4af7ba585b6fecae010b64051556d5d9b11
The arch/ directory as made a submodule with e9a3cc34576f3942b7bbd3d5fecf697cde7748eb

The normal directories were restored sometime after 1645c75d4b0ae8f8c375b9d77a2e84c28ef13f45 (last submodule update).

acassis, 2023-12-27:
Thank you @patacongo for confirming the issue. At least for AVR issue, using version 7.16 to 7.17 is enough to track the serial issue.

patacongo, 2023-12-27:
> Thank you @patacongo for confirming the issue. At least for AVR issue, using version 7.16 to 7.17 is enough to track the serial issue.

What I don't understand is why github won't access those directories.  There must be something there it cannot handle.  It worked well on BitBucket (where the change was made) and the files are clearly in the repository because you can see the file history for arch/, Documentation/, and configs all the way back to the beginning.

github is not tolerant of any quirkiness in the repository.  It won't access anything it doesn't like.  For example if you look at the contributors:  https://github.com/apache/nuttx/graphs/contributors , it doesn't see anything before about 2014.  That is the date when the repository was converted from SVN to GIT.  So the statistics miss something like 20 thousand contributions.

My belief is that that is caused by the fake email addresses that the old SVN-to-GIT conversion tool created.  Ijust use a UUID for the email address.

Bitbucket can also see all commits from day one when NuttX was under CVS, then, SVN, then GIT.

acassis, 2023-12-27:
@patacongo I think it is not much problematic since nobody it too worried about preserving all arches and boards, but I think it should be nice to have the complete history of NuttX saved in the git.

Maybe if you have the original bitbucket, it should be nice to keep it as Read-Only with a comment at top saying that the development is happening here in the github (this repository is preserved only to history purposes).