apache/nuttx #11452: Toolchain command do not take effect in CMake build after menuconfig settings

xuxin930, 2023-12-26:
Here is an example:

Our current Toolchain File overwrite the ARCHIVE command
https://github.com/apache/nuttx/blob/992719730f355ed3132c3204153f919822870074/arch/arm/src/cmake/Toolchain.cmake#L92-L94
but `<CMAKE_AR>` makes different choices depending on whether `LTO` is enabled or not.
https://github.com/apache/nuttx/blob/992719730f355ed3132c3204153f919822870074/arch/arm/src/cmake/Toolchain.cmake#L75-L86
If everything is configured in defconfig, everything will work normally.
But if `LTO` is **not** enabled in defconfig and I call `cmake --build -t menuconfig` to open `LTO`, then the expectation is **different** from the setting.

Because the execution of CMake is divided into three stages: **_configure_**, **_generate_** and **_build_**.
the Toolchain command (such as CMAKE_AR above) is configured in stages  **_configure_**, **_generate_**  and is solidified in the generated build file (such as Makefile).
so in the end this menuconfig modification will NOT take effect.

acassis, 2023-12-28:
@pkarashchenko any idea?

xiaoxiang781216, 2023-12-28:
In this case, we should regenerate Makefile?

casaroli, 2026-11-08:
Possibly fixed by #20816 and #20085.

closed 2026-11-08 by casaroli