apache/nuttx #11462: Fix arm-none-eabi-ld: warning: /github/workspace/sources/nuttx/nuttx has a LOAD segment with RWX permissions

xiaoxiang781216, 2023-12-28:
Report from all ci console, like this:
https://github.com/apache/nuttx/actions/runs/7347758008/job/20004759931

pkarashchenko, 2023-12-28:
I'm not sure how to fix this especially when the part of the code is relocated to SRAM.

codusnocturnus, 2024-02-04:
Just getting into this level of development, and am only working with STM32, so I can't say it's an all-around fix with no edge cases, but the solution on this blog seems to work: [Solution to LOAD segment with RWX permissions warning with CMSIS](https://metebalci.com/blog/solution-to-load-segment-with-rwx-permissions-warning-with-cmsis)

To summarize, define a PHDRS block in the linker script with descriptions for the text, data, and bss segments, and apply the descriptions to the related output sections.

The warning goes away, and the ELF listing will show that the flags on segment 0 are now "correct".

casaroli, 2026-11-08:
Fixed by #14474 and #16886.

closed 2026-11-08 by casaroli