apache/nuttx #11463: NSH: ps command broken between release 12.0 and 12.1

maxikrie, 2023-12-28:
I noticed, that the ps command of the NSH library broke between release 12.0 and 12.1. 

Specifically, I am testing the nucleo-h743zi2:nsh config, where I enabled CONFIG_FS_PROCFS and CONFIG_NSH_ARCHINIT to get the ps command in NSH. When executing the command (sometimes only on the second try) I get a hardfault. This starts happening with release 12.1 (still apparent with 12.3) and works as expected on 12.0.

Before debugging this in more detail. maybe somebody can help pointing in the right direction? I see that the LR points to tls_get_info and the PC to getumask during the hardfault.

Thanks!

acassis, 2023-12-29:
@maxikrie I faced similar issue helping @JorgeGzm with witte board powered by stm32h753bi. I think it could be some issues related to STM32H7, because in other boards the issues doesn't happen

maxikrie, 2023-12-29:
I tracked it down to changes introduced in nsh_console.c, nsh_console.h, nsh_proccmds.c between 12.0 and 12.1. If I revert only these files to 12.0, I don't see this issue.

I haven't reviewed these files in detail, but isn't the nsh library hardware agnostic? So how could this be STM32H7 specific issue?

acassis, 2023-12-29:
@maxikrie so maybe this is not the same issue we are seeing in our board.

Confirming that ps on stm32f4discovery still working:

```
nsh> uname -a
NuttX 12.3.0 ffc7a192a0-dirty Dec 23 2023 17:58:59 arm stm32f4discovery

nsh> free
                   total       used       free    maxused    maxfree  nused  nfree
        Umem:     193400       7832     185568       9336     125008     37      3

nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK           STACK COMMAND
    0     0   0 FIFO     Kthread   - Ready              0000000000000000 001008 Idle_Task
    2     2 100 RR       Task      - Running            0000000000000000 002008 nsh_main
nsh>
```

tmedicci, 2023-12-30:
Hi!

PTAL at https://github.com/apache/nuttx/pull/11425

On ESP32-S2, the problem with the `ps` command was related to the NSH's task stack size. The error didn't occur every time the command was run, but eventually. Increasing the stack size fixed the problem. My guess: TCB increased and requires more space depending on the arch.

maxikrie, 2023-12-30:
Yes, I noticed this as well. 

So, I increased the stack size for NSH task 
`CONFIG_SYSTEM_NSH_STACKSIZE=4096`
but I also had to increase the stack size of the init task
`CONFIG_INIT_STACKSIZE=4096`

Does this make any sense or is this simply obfuscating the problem? In any case, so far I am not witnessing any crashes.

acassis, 2024-01-01:
@maxikrie good question, I think more investigation will be needed. Note that I'm using the default stacksize in stm32f4discovery board.

maxikrie, 2026-10-26:
@acassis I found the cause: it's a stack overflow of the task running NSH, not an H7 bug. With `CONFIG_INIT_ENTRYPOINT="nsh_main"` NSH runs as the init task, so its stack is `CONFIG_INIT_STACKSIZE` (2048), not `CONFIG_SYSTEM_NSH_STACKSIZE`. The H7 Nucleo `nsh` configs set `CONFIG_DEBUG_SYMBOLS=y`, which makes `-O0` (`DEBUG_NOOPT`) the default optimization, and at `-O0` `ps` needs more than 2048 bytes.

Measured on a NUCLEO-H753ZI, master b9d1d6a834, `nucleo-h753zi:nsh` + `FS_PROCFS`, `STACK_COLORATION`, `DEBUG_ASSERTIONS`, 200 × `ps`:

| build | result | NSH stack used (ps) |
|---|---|---|
| defconfig (-O0), INIT_STACKSIZE 2048 | first `ps` crashes: `sp_out_of_range: ERROR: Stack pointer 38000ed8 is not within the stack`, fault in `nxsig_pendingset()` | — |
| -O0, INIT_STACKSIZE 4096 | 200/200 ok | 2324 bytes |
| -Os (`DEBUG_FULLOPT`), INIT_STACKSIZE 2048 | 200/200 ok | 1744 bytes (87 %) |
| nRF52840-DK `nrf52840-dk:nsh` (-Os), 2048 | 50/50 ok | 1440 bytes |

So that's why `INIT_STACKSIZE=4096` fixed it for me (and a bigger NSH stack for tmedicci on the ESP32-S2), and why other boards (optimized builds) don't show it. Should the H7 `nsh` configs get a bigger `CONFIG_INIT_STACKSIZE` (or drop `DEBUG_SYMBOLS`)? Otherwise I'd close this.

maxikrie, 2026-10-30:
@acassis ping (or should we include anybody else?)

Following up on my measurements above: the crash is explained by the init task's stack overflowing when it runs nsh_main. Increasing CONFIG_INIT_STACKSIZE to 4096 passed 200 consecutive ps calls on the H753; an optimized build also passed with 2048.

I think we can close this as diagnosed. Increasing the stack in the affected H7 board configs can be handled separately.

xiaoxiang781216, 2026-10-30:
Thanks for verification, @maxikrie .

closed 2026-10-30 by xiaoxiang781216