apache/nuttx #11471: ESP32S3: WIFI Soft-AP Mode does not compatiable with SMP

w2016561536, 2024-01-02:
## Problem:  
When I tried to use my phone to join the Soft-AP from my board. The terminal got stuck. I cannot input anything to terminal, and the access point will disappear in a few secounds. However everything goes well if i turn down the SMP. 

**Base Config** : esp32s3-devkit:sta_softap  
**Chages** : Enable SMP, set NCPUS to 2. (Through menuconfig)  
**Test methods** : Use Xshell to connect to NUTTX-Shell through UART0,  input the commands 

`  ifup wlan1  `
`  dhcpd_start wlan1  `
`  wapi psk wlan1 12345678 3  `
`  wapi essid wlan1 nuttxap 1  `
 
Then, connect to the AP. And it happens.

acassis, 2024-01-02:
Hi @w2016561536, thanks for reporting this issue. It is strange because everything seems to work fine in ESP32.

@tmedicci did you see this issue before?

tmedicci, 2024-01-02:
> Hi @w2016561536, thanks for reporting this issue. It is strange because everything seems to work fine in ESP32.
> 
> @tmedicci did you see this issue before?

Yes, it's a known issue. The solution was already proposed internally but I'm on leave and I had no feedback on the proposed solution. I'll take a look as soon as I return to work next week. Please keep the issue open until it's properly fixed.

w2016561536, 2024-01-03:
it's very glad to get feedback so fast , thanks!

tmedicci, 2024-01-10:
> it's very glad to get feedback so fast , thanks!

Hi! Could you please test [this](https://github.com/apache/nuttx/pull/11519) MR?

closed 2024-01-11 by masayuki2009

w2016561536, 2024-01-12:
> > it's very glad to get feedback so fast , thanks!
> 
> Hi! Could you please test [this](https://github.com/apache/nuttx/pull/11519) MR?

Sorry for delayed reply. It works perfectly! Thanks everyone's efforts to NuttX!