apache/nuttx #11473: cmake build seems lacking sethost.sh equivalent

yamt, 2024-01-04:


xuxin930, 2024-01-05:
I think this should have been designed in the nuttx_kconfig module. 
https://github.com/apache/nuttx/blob/3bac4492865ad3bd4c293e9d751163e090634678/cmake/nuttx_kconfig.cmake#L68
current settings are not perfect enough.

yamt, 2024-01-05:
> I think this should have been designed in the nuttx_kconfig module.
> 
> https://github.com/apache/nuttx/blob/3bac4492865ad3bd4c293e9d751163e090634678/cmake/nuttx_kconfig.cmake#L68
> 
> current settings are not perfect enough.

thank you. i was not aware of the cmake fragment.

however, i think we need something actually involving kconfig-tweak
because some other configs depend on these CONFIG_HOST_xxx.
eg. https://github.com/apache/nuttx/blob/3bac4492865ad3bd4c293e9d751163e090634678/arch/sim/Kconfig#L210

anchao, 2024-01-11:
kconfiglib provides a method named `setconfig`, which similar to `kconfig-tweak`, I think we can add the cmake target or command of `setconfig` wrapper to support the sethost feature:
https://github.com/ulfalizer/Kconfiglib/blob/master/setconfig.py

W-Mai, 2024-01-27:
I wrote a cmake `sethost`, but it's a little hacky.

Now that I see this discussion, I decided to follow this discussion and try to modify my code.

- #11608

anchao, 2024-01-27:
> I wrote a cmake `sethost`, but it's a little hacky.
> 
> 
> 
> Now that I see this discussion, I decided to follow this discussion and try to modify my code.
> 
> 
> 
> - #11608

It's best to align with @xuxin930,maybe he's working on the relevant implementation

W-Mai, 2024-01-27:
> It's best to align with @xuxin930,maybe he's working on the relevant implementation

Great!

casaroli, 2026-11-08:
Possibly fixed by #13175.