apache/nuttx #11476: ESP32S3:  LEDC does not have output.

w2016561536, 2024-01-04:
## Problem:   
Use the examples of pwm in nuttx. Execute the command ` pwm ` in terminal. The output is  always LOW (GPIO 2, the default pin of channel 0). And I'm sure the hardware is OK because it can works in idf. More accurately, the GPIO2 is floating after reset, and connect to ground after execute the command.  

## Test conditions:    
Configs: ` esp32s3-devkit:pwm `,  no changes.  
Hardware: ESP32S3-WROOM-1(M0N4R2)  

## More tests: 
It seems that ` PWM Mutiple Outpur Channels Support ` does not open. So I open it and configure the examples of pwm channel to 0 and 1 (default is 1 and 2). Test again. Nothing changed, except GPIO 3 output low either.

acassis, 2024-01-05:
Hi @w2016561536 thank you for reporting this issue, when I tested it last year everything was working fine. @tmedicci do you know if it stopped to work?

tmedicci, 2024-01-08:
I'll take a look soon. I returned from my vacation today and there is a lot of stuff to get back to the routine. I'm sorry for that.

w2016561536, 2024-01-16:
> I'll take a look soon. I returned from my vacation today and there is a lot of stuff to get back to the routine. I'm sorry for that.

I have found the problem of ledc does not have the output. I am trying to make it better to fit for a wider range of frequency. Thanks for your contributions!

PR #11557 (2024-01-18): esp32s3/pwm: Fix pwm output
## Summary
1. Fix pwm output always low problem.  
2. Add multi channel support in defconfig  

## Impact
Solve the problem of LEDC output low and close #11476  

## Testing
Hardware: esp32s3-wroom-1 (M0N4R2)  
[image]
[image]
[image]
closed 2024-01-18 by acassis