apache/nuttx #11490: Schedule policy for stm32f4discovery(stm32f407)

nopnop2002, 2024-01-05:
I'm using Nuttx V10.2.
I believe nutxx's standard scheduling policy is FIFO.
However, with STM32F407, if two tasks with the same priority are started, the RR scheduling policy will be applied.

This is that logging.
Two tasks with the same priority started running on 4808Tick.
```
task_create name:myTask1 priority:100
task_create name:myTask2 priority:100
myTask1 start PID:12 loop:50000000 wait:0 system_timer:4808
myTask2 start PID:13 loop:50000000 wait:0 system_timer:4808
nsh> ps
  PID PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK   STACK COMMAND
    0   0 FIFO     Kthread N-- Ready              00000000 001000 Idle Task
    2 100 RR       Task    --- Running            00000000 002032 init
   12 100 RR       Task    --- Ready              00000000 002008 myTask1 50000000 0
   13 100 RR       Task    --- Ready              00000000 002008 myTask2 50000000 0
nsh> myTask1 end PID:12 system_timer:5336
myTask2 end PID:13 system_timer:5344
```

acassis, 2024-01-05:
@nopnop2002 by default RR is select to this board:
```
CONFIG_RR_INTERVAL=200
```

If you want to disable it set the value to 0:
```
CONFIG_RR_INTERVAL=0
```

Please consider helping, adding the documentation to make the scheduling policy FIFO vs RR clear.

Could you please close this issue? (since it is a configuration detail)

acassis, 2024-01-05:
@xiaoxiang781216 @patacongo @pkarashchenko I think it the default config is prone to error.
It should be better to have CONFIG_SCHED_FIFO, CONFIG_SCHED_RR, CONFIG_SCHED_SPORADIC, etc

What do you think?

nopnop2002, 2024-01-05:
@acassis 

I changed the CONFIG_RR_INTERVAL value to 0 and built again.

But nothing changes.

```
nsh> task_test test2
task_create name:myTask1 priority:100
task_create name:myTask2 priority:100
myTask1 start PID:13 loop:50000000 wait:0 system_timer:24186
myTask2 start PID:14 loop:50000000 wait:0 system_timer:24186
nsh> myTask1 end PID:13 system_timer:24454
myTask2 end PID:14 system_timer:24722
```

acassis, 2024-01-06:
@nopnop2002 thank you for pointing it, so our documentation is incorrect. If you can help, please modify the right documentation file at nuttx/Documentation/ and submit a PR to fix it.

BTW, I will reopen it, because setting RR_INTERVAL to 0 fix the issue, but the issue is deeper (documentation, better Kconfig symbol usage, etc)