apache/nuttx #11502: version.h for version 12.4 is incorrect

nopnop2002, 2024-01-08:
I updated from version 12.3 to version 12.4.
When we build, version.h is automatically generated.
However, the contents of version.h are version 12.3.

```
$ make
Create version.h
LN: platform/board to /home/nop/nuttxspace/apps/platform/dummy
Register: hello
Register: nsh
Register: sh

$ cat $HOME/nuttxspace/nuttx/include/nuttx/version.h
/* version.h -- Autogenerated! Do not edit. */

#ifndef __INCLUDE_NUTTX_VERSION_H
#define __INCLUDE_NUTTX_VERSION_H

#define CONFIG_VERSION_STRING "12.3.0"
#define CONFIG_VERSION_MAJOR 12
#define CONFIG_VERSION_MINOR 3
#define CONFIG_VERSION_PATCH 0
#define CONFIG_VERSION_BUILD "9852428953"

#define CONFIG_VERSION ((CONFIG_VERSION_MAJOR << 16) |\
                        (CONFIG_VERSION_MINOR << 8) |\
                        (CONFIG_VERSION_PATCH))
```

jerpelea, 2024-01-08:
12.4 release is WIP.

chenzhihong007, 2024-01-26:
> 12.4 release is WIP.

Hi Jerpelea. May I ask when the official version of 12.4 is expected to be released? Thank you!