apache/nuttx #11503: raspberrypi-pico/usbnsh: it's not work, if we add a usb-hub between the pico&pc

damon-yun, 2024-01-08:
>./tools/configure.sh -l raspberrypi-pico:usbnsh
>make -j8
then test it , it's not work, if we add a usb-hub between the pico&pc. remove the usb-hub connect direct will be good. but we need it.

acassis, 2024-01-08:
@damon-yun I think the issue could be related to how to host OS is detecting NuttX CDC/ACM device. Probably you will need to use a USB Sniffer (like this one: https://github.com/ataradov/usb-sniffer ) to investigate what is happening in the bus.

damon-yun, 2024-01-16:
@acassis Unfortunately I didn't make this tool, but I compared it with i.mx rt & stm32h7.  All work good with Win11 & Ubuntu 22.04, except raspberrypi-pico. Maybe we can wait for feedback from other user.