apache/nuttx #11529: Nuttx ./tools/configure.sh -l stm32f103-minimum/userled does not blink led.

mxrtos, 2024-01-11:
Install nuttx and the stm32 toolchain. (You can follow Alan nuttx channel tutorials)
Run:
```
cd /home/juan/nuttxspace/nuttx
./tools/configure.sh -l stm32f103-minimum/userled
make -j4
openocd -f interface/stlink-v2.cfg -f target/stm32f1x.cfg -c init -c "reset halt" -c "flash write_image erase nuttx.bin 0x08000000"
picocom -b 115200  /dev/ttyUSB0
```
Then in nutshell press enter:
```
nsh> ?
nsh>
```
The leds program is not there, the board does not blink the led. All settings were verified in "make menuconfig"

acassis, 2024-01-12:
Hi @mxrtos, could you please verify if you see "Register: leds" printed when you compile the code, like this:
```$ make -j
Create version.h
LN: platform/board to /home/alan/nuttxspace/apps/platform/dummy
Register: leds
Register: nsh
Register: sh
chip/stm32_gpio.c:41:11: note: '#pragma message: CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py'
   41 | #  pragma message "CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py"
      |           ^~~~~~~
CPP:  /home/alan/nuttxspace/nuttx/boards/arm/stm32/stm32f103-minimum/scripts/ld.script-> /home/alan/nuttxspace/nuttx/boards/arm/stm32/stm32f103-minimum/scripts/LD: nuttx
CP: nuttx.bin
```

I think there could be some issues in your host system

mxrtos, 2024-01-12:
Sure!
I put a bit <-------------------------------------------------------------------------------------------- in that line:

```
➜  nuttx git:(master) make distclean
➜  nuttx git:(master) ./tools/configure.sh -l stm32f103-minimum/userled
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
CP: arch/dummy/Kconfig to /home/juan/nuttxspace/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/juan/nuttxspace/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/juan/nuttxspace/apps/platform/dummy
LN: include/arch to arch/arm/include
LN: include/arch/board to /home/juan/nuttxspace/nuttx/boards/arm/stm32/stm32f103-minimum/include
LN: drivers/platform to /home/juan/nuttxspace/nuttx/drivers/dummy
LN: include/arch/chip to /home/juan/nuttxspace/nuttx/arch/arm/include/stm32
LN: arch/arm/src/chip to /home/juan/nuttxspace/nuttx/arch/arm/src/stm32
LN: arch/arm/src/board to /home/juan/nuttxspace/nuttx/boards/arm/stm32/stm32f103-minimum/../common
LN: arch/arm/src/board/board to /home/juan/nuttxspace/nuttx/boards/arm/stm32/stm32f103-minimum/src
mkkconfig in /home/juan/nuttxspace/apps/audioutils
mkkconfig in /home/juan/nuttxspace/apps/benchmarks
mkkconfig in /home/juan/nuttxspace/apps/boot
mkkconfig in /home/juan/nuttxspace/apps/canutils
mkkconfig in /home/juan/nuttxspace/apps/crypto
mkkconfig in /home/juan/nuttxspace/apps/examples/mcuboot
mkkconfig in /home/juan/nuttxspace/apps/examples
mkkconfig in /home/juan/nuttxspace/apps/fsutils
mkkconfig in /home/juan/nuttxspace/apps/games
mkkconfig in /home/juan/nuttxspace/apps/graphics
mkkconfig in /home/juan/nuttxspace/apps/industry
mkkconfig in /home/juan/nuttxspace/apps/inertial
mkkconfig in /home/juan/nuttxspace/apps/interpreters/luamodules
mkkconfig in /home/juan/nuttxspace/apps/interpreters
mkkconfig in /home/juan/nuttxspace/apps/logging
mkkconfig in /home/juan/nuttxspace/apps/lte
mkkconfig in /home/juan/nuttxspace/apps/math
mkkconfig in /home/juan/nuttxspace/apps/mlearning
mkkconfig in /home/juan/nuttxspace/apps/netutils
mkkconfig in /home/juan/nuttxspace/apps/sdr
mkkconfig in /home/juan/nuttxspace/apps/system
mkkconfig in /home/juan/nuttxspace/apps/testing
mkkconfig in /home/juan/nuttxspace/apps/wireless/bluetooth
mkkconfig in /home/juan/nuttxspace/apps/wireless/ieee802154
mkkconfig in /home/juan/nuttxspace/apps/wireless
mkkconfig in /home/juan/nuttxspace/apps
#
# configuration written to .config
#
➜  nuttx git:(master) make
Create version.h
LN: platform/board to /home/juan/nuttxspace/apps/platform/dummy
Register: leds      <--------------------------------------------------------------------------------------------
Register: nsh
Register: sh
CC:  chip/stm32_gpio.c chip/stm32_gpio.c:41:11: note: '#pragma message: CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py'
   41 | #  pragma message "CONFIG_STM32_USE_LEGACY_PINMAP will be deprecated migrate board.h see tools/stm32_pinmap_tool.py"
      |           ^~~~~~~
CPP:  /home/juan/nuttxspace/nuttx/boards/arm/stm32/stm32f103-minimum/scripts/ld.script-> /home/juan/nuttxspace/nuttx/boards/arm/stm32/stm32f103-minimum/scriLD: nuttx
CP: nuttx.bin
```

Thanks!

mxrtos, 2024-01-12:
I will continue to check this, I have tried in Kubuntu, and Manajro VMs so far.