apache/nuttx #11532: posix signal use

xiaotailang, 2024-01-12:
I am trying to use the posix signal interface on the stm32. After opening the configuration items SIG_DEFAULT and CONFIG_SIG_SIGKILL_ACTION, the system will be stuck at the memcpy((uint32_t *)regs[REG_R1], regulators, XCPTCONTEXT_SIZE); in the up_svcall function of the up_svcall.c file during system startup. Is there an additional configuration item that I have missed if I want to support signal?

alandeassis, 2024-01-14:
@xiaotailang depeding when you are trying to use it, the POSIX functions are not initialized yet and will not work. Could you please show some generic example to let us to test it? If you can try also in the SIMulator, it will be easier to reproduce

xiaotailang, 2024-01-16:
during debugging, I discovered that the issue was caused by a conflict between a custom signal I defined and the system's native signals. After modifying my custom signal, everything works as expected.

alandeassis, 2024-01-16:
Hi @xiaotailang please give more details, it could help other people facing similar issue in the future. Maybe it could be in our documentation, in FAQ or even Guide.

xiaotailang, 2024-01-24:
Hi@alandeassis，The values of the two signals sig_usr1 and sig_usr2. I made a mistake in writing one of them, which caused a problem. I tested using the system default values and there was no problem.

closed 2024-02-16 by xiaoxiang781216