apache/nuttx #11539: ESP32S3:  SPI cannot use.

w2016561536, 2024-01-13:
## Problem:  
Use system spitool, the spi bus cannot be got.  
[image]  

## Config:
``` #
# This file is autogenerated: PLEASE DO NOT EDIT IT.
#
# You can use "make menuconfig" to make any modifications to the installed .config file.
# You can then do "make savedefconfig" to generate a new defconfig file that includes your
# modifications.
#
# CONFIG_ARCH_LEDS is not set
# CONFIG_NSH_ARGCAT is not set
# CONFIG_NSH_CMDOPT_HEXDUMP is not set
CONFIG_ARCH="xtensa"
CONFIG_ARCH_BOARD="esp32s3-devkit"
CONFIG_ARCH_BOARD_COMMON=y
CONFIG_ARCH_BOARD_ESP32S3_DEVKIT=y
CONFIG_ARCH_CHIP="esp32s3"
CONFIG_ARCH_CHIP_ESP32S3=y
CONFIG_ARCH_CHIP_ESP32S3WROOM1=y
CONFIG_ARCH_INTERRUPTSTACK=2048
CONFIG_ARCH_STACKDUMP=y
CONFIG_ARCH_XTENSA=y
CONFIG_BOARD_LOOPSPERMSEC=16717
CONFIG_BUILTIN=y
CONFIG_DEBUG_ASSERTIONS=y
CONFIG_DEBUG_FEATURES=y
CONFIG_ESP32S3_I2C0=y
CONFIG_ESP32S3_I2C0_SCLPIN=6
CONFIG_ESP32S3_I2C0_SDAPIN=7
CONFIG_ESP32S3_I2C1=y
CONFIG_ESP32S3_I2C1_SCLPIN=15
CONFIG_ESP32S3_I2C1_SDAPIN=16
CONFIG_ESP32S3_SPI2=y
CONFIG_ESP32S3_SPI2_CLKPIN=2
CONFIG_ESP32S3_SPI2_CSPIN=1
CONFIG_ESP32S3_SPI2_MISOPIN=41
CONFIG_ESP32S3_SPI2_MOSIPIN=42
CONFIG_ESP32S3_SPI3=y
CONFIG_ESP32S3_SPI3_CLKPIN=13
CONFIG_ESP32S3_SPI3_CSPIN=11
CONFIG_ESP32S3_SPI3_MISOPIN=12
CONFIG_ESP32S3_SPI3_MOSIPIN=14
CONFIG_ESP32S3_SPI_DMA=y
CONFIG_ESP32S3_UART0=y
CONFIG_FS_PROCFS=y
CONFIG_HAVE_CXX=y
CONFIG_HAVE_CXXINITIALIZE=y
CONFIG_I2CTOOL_DEFFREQ=100000
CONFIG_I2CTOOL_MAXBUS=1
CONFIG_I2C_RESET=y
CONFIG_IDLETHREAD_STACKSIZE=3072
CONFIG_INIT_ENTRYPOINT="nsh_main"
CONFIG_INTELHEX_BINARY=y
CONFIG_NSH_ARCHINIT=y
CONFIG_NSH_BUILTIN_APPS=y
CONFIG_NSH_FILEIOSIZE=512
CONFIG_NSH_LINELEN=64
CONFIG_NSH_READLINE=y
CONFIG_PREALLOC_TIMERS=4
CONFIG_RAM_SIZE=114688
CONFIG_RAM_START=0x20000000
CONFIG_RR_INTERVAL=200
CONFIG_SCHED_WAITPID=y
CONFIG_SMP=y
CONFIG_SMP_NCPUS=2
CONFIG_SPITOOL_MINBUS=2
CONFIG_START_DAY=28
CONFIG_START_MONTH=6
CONFIG_START_YEAR=2022
CONFIG_SYSLOG_BUFFER=y
CONFIG_SYSTEM_I2CTOOL=y
CONFIG_SYSTEM_NSH=y
CONFIG_SYSTEM_SPITOOL=y
CONFIG_UART0_SERIAL_CONSOLE=y
```

## Hardware :
Module: ESP32-S3-WROOM-1(M0N4R2)

alandeassis, 2024-01-14:
@w2016561536 this issue is happening because you are not registering the SPI char device, you need to call spi_register() to do it. Please look how it is done in other boards.

w2016561536, 2024-01-14:
> @w2016561536 this issue is happening because you are not registering the SPI char device, you need to call spi_register() to do it. Please look how it is done in other boards.

Thanks for your advice. I have found SPI confilect bug in ESP32S3. I will open a pr soon.

PR #11554 (2024-01-17): esp32s3/spi: Add SPI bus init in bringup and fix SPI bus 2 and 3 conflict
## Summary
1. Add spi bus init in esp32s3_bringup.c
2. Fix IOMUX conflict between spi bus 2 and 3
3. Add spi defconfig

## Impact
Fix esp32s3 spi bus and close #11539 

## Testing
[image]
Spi bus 2 and 3 mount icm-20602 and FM25V05 respectively. Use spi-tool in cli to read their id.
closed 2024-01-17 by tmedicci