apache/nuttx #11555: Question about rpmsg_register_callback()

hzc04, 2024-01-16:
Hello, I've been reading the code related to rptun recently, and I have a question about the implementation of rpmsg_register_callback().

I think after creating the rpmsg_device, I need to register a rpmsg service using rpmsg_register_callback(). This service will be added to g_rptun_cb.
Until the remote send a ns msg to create rpmsg channel, the host will scan the g_rptun_cb and call ns_bind() to bind a rpmsg_endpoint. This endpoint will be added to priv->bind. After that, we can communicate normally, and see what endpoints have been bound based on priv->bind.

That's my current understanding. I'm not sure I've got it right.

I don't understand why, in the process of rpmsg_register_callback(), if it finds that this cb(or endpoint?) is already bound, it removes the cb from priv->bind and triggers ns_bind()..? (https://github.com/apache/nuttx/blob/master/drivers/rptun/rptun.c#L1111) Don't need to trigger ns_bind() via ns msg?

Can you please correct my understanding? Thank you.

Would you please explain this to me? Thanks a lot!

xiaoxiang781216, 2024-01-17:
https://github.com/apache/nuttx/blob/master/drivers/rptun/rptun.c#L1111 is invalid, please provide the correct location.

hzc04, 2024-01-17:
> https://github.com/apache/nuttx/blob/master/drivers/rptun/rptun.c#L1111 is invalid, please provide the correct location.

Yesterday this code was split into rpmsg.c

https://github.com/apache/nuttx/blob/master/drivers/rpmsg/rpmsg.c#L184

xiaoxiang781216, 2024-01-17:
Since the ns message may come before you call rpmsg_register_callback, the framework has to append the endpoint to bind list in this case, and notify you later. After this point, it isn't used anymore and then removed.

hzc04, 2024-01-17:
> Since the ns message may come before you call rpmsg_register_callback, the framework has to append the endpoint to bind list in this case, and notify you later. After this point, it isn't used anymore and then removed.

I clearly understand this now.  Many thanks!!

closed 2024-01-17 by hzc04