apache/nuttx #11567: esp32s3/smp: Dual Core Scheduling Problem.

w2016561536, 2024-01-19:
At first, thank you for reading this issue. I'm not quite sure if it occurred by NuttX scheduling. Perhaps applications cause this issue.

[image]
 Here is the process information of my board. It is running PX4-Autopilot, which is based on NuttX. On CPU0, there is one task taking up all the resources. And no new task can be scheduled on CPU0 (if I run smp test at this moment). Howerver this application has no problem on single core. The process information below.
[image]

And what makes me puzzled that ALWAYS CPU0 happens this problem and it never happens on CPU1. And, the task stuck on the CPU0 is random. It could change after reset. But sometimes the app runs well and no task stuck on one core.

tmedicci, 2024-01-19:
HI @w2016561536 !

I didn't get exactly what is your problem, but I saw no problems regarding your process scheduling in the images you shared. Could you please elaborate what is the problem?

w2016561536, 2024-01-19:
In picture 1, wq:rate_ctrl take up all the cpu resources. But actually, it does not run, just stuck there. It look like cpu 0 is dead

tmedicci, 2024-01-19:
> In picture 1, wq:rate_ctrl take up all the cpu resources. But actually, it does not run, just stuck there. It look like cpu 0 is dead

The picture does not show that. It shows that `wq:rate_ctrl` takes 50.2% of the CPU time. It is expected it to be that high because this task takes the highest priority (perhaps you should lower it). About being stuck: you would need to analyze the app itself: try to attach the debugger and check where it is being stuck.

w2016561536, 2024-01-19:
> The picture does not show that. It shows that `wq:rate_ctrl` takes 50.2% of the CPU time. It is expected it to be that high because this task takes the highest priority (perhaps you should lower [...]

In NuttX，when dual core is enabled and system has very low load, each Idle task takes 50% time. Thank you for your advice. I will use debugger to inspect what happens on Core 0 next day. Here is middle night. I need to go to bed. Have a nice day. @tmedicci

acassis, 2024-01-20:
@masayuki2009 since you did many improvements over original SMP, maybe you can have some suggestion to help here.

masayuki2009, 2024-01-22:
@acassis 
> @masayuki2009 since you did many improvements over original SMP, maybe you can have some suggestion to help here.

I think @tmedicci's comments are correct.
However, we should clarify which version of the nuttx  @w2016561536 is using.
Also, we should clarify the defconfig (i.e. esp32s3-devkit:smp) that @w2016561536 modified.

FelipeMdeO, 2026-08-17:
@cederom , let's follow this thread for more some days, but I think this issue can be closed.
Any comment @w2016561536 ?

cederom, 2026-08-18:
Looks like no more traffic here since 2024, lets close and reopen if needed, thank you everyone involved :-)

closed 2026-08-18 by cederom