apache/nuttx #11586: Raspberry pi 4 model B support

TaiJuWu, 2024-01-22:
Can Nuttx support raspberry 4 model B?
Or anyone work on this board?

alandeassis, 2024-01-23:
@TaiJuWu the initial support to Raspberry Pi board (the first board) was removed by Greg some years ago because nobody was interested to still working in the port: https://bitbucket.org/patacongo/obsoleted/src/master/Patches/Remove-arch-bcm2708-2019-01-02.patch

If you want to work on it, you can look that initial support and use it as reference.

TaiJuWu, 2024-01-23:
> @TaiJuWu the initial support to Raspberry Pi board (the first board) was removed by Greg some years ago because nobody was interested to still working in the port: [...]

Thanks for your information.

linguini1, 2024-06-28:
Checking if this is in development at all? I am interested in contributing to get support for this board.

TaiJuWu, 2024-06-29:
> Checking if this is in development at all? I am interested in contributing to get support for this board.

No, I am not working on it.

linguini1, 2024-07-23:
Could this issue be assigned to me? I have begun work on it.

xiaoxiang781216, 2024-07-23:
> Could this issue be assigned to me? I have begun work on it.

Done, thanks.

linguini1, 2024-08-07:
So I have an initial implementation for this board (the 4GB variant is all I have access to for testing right now) which boots successfully and has NSH running with a UART console. It passes the `ostest` as well.

Obviously I plan to implement some drivers for the board so it can actually be used for programming/development (I2C, remaining UARTs, GPIO, etc.). I am wondering, however, at what point is a port to a new board considered "good enough" to be merged? I imagine that full support for everything on the board (like HDMI ports, Bluetooth, etc.) does not have to be implemented to be merged, but is there a guideline on how complete it must be at a minimum to become part of the main source tree?

I ask only because I think that at some point in the development process I'd like to be able to call it "complete enough" to allow for it to be merged and more accessible to others to make changes/add features.

xiaoxiang781216, 2024-08-07:
it's enough to implement uart and timer driver to ensure the port enter nsh correctly.

csukuangfj, 2024-12-13:
Any progress on this issue?

linguini1, 2024-12-13:
> Any progress on this issue?

@csukuangfj I have managed to get NuttX booting into NSH on the 4GB RAM model, haven't been able to test on other models. I have a partially made I2C driver which is quite buggy and is only able to do read operations on the bus. I was starting on SPI aand got quite busy with my coursework. What I'll likely do is open a PR with what I've got working so far and mark features as incomplete on the board docs. Then hopefully the community can pitch in and we can get more features done quicker!

PR #15188 (2024-12-17): Add support for the BCM2711
## Summary

Closes #11586. 

This PR adds semi-functional support for the BCM2711 and Raspberry Pi 4B. The working features are a Mini-UART shell which can boot into NSH. I have run OStest and confirmed it works. The latest changes also boot.

## Impact

Although the feature implementation for this board is minimal right now, I hope that by releasing the work I've done so far it is easier for other community members to start adding drivers/features to the board's support on NuttX. Several people were interested in using the Pi 4B with NuttX and it's a reasonably pop [... cut]
closed 2024-12-17 by xiaoxiang781216