apache/nuttx #11611: size_report.py  does not work with some versions "anytree" package

dbnnrmn, 2024-01-28:
The report script does not have "any tree" package in dependencies and need some old versions of this package. Version 2.12.1 falls with this error: 

```
File "/mnt/sdb1/nuttxspace/nuttx/./tools/size_report.py", line 815, in <module>
    main()
  File "/mnt/sdb1/nuttxspace/nuttx/./tools/size_report.py", line 802, in main
    root = generate_any_tree(symbol_dict, symsize, common_path_prefix)
  File "/mnt/sdb1/nuttxspace/nuttx/./tools/size_report.py", line 559, in generate_any_tree
    root = TreeNode("Root", "root")
  File "/mnt/sdb1/nuttxspace/nuttx/./tools/size_report.py", line 533, in __init__
    self.size = size
AttributeError: can't set attribute 'size'

```

But version 2.8.0 works fine

acassis, 2024-01-28:
@xiaoxiang781216 any idea about it? I never saw this error on Ubuntu!

TaiJuWu, 2024-04-28:
Maybe you can see this PR #10879

maxikrie, 2026-11-11:
@xiaoxiang781216 @acassis This is fixed on master by e0e6e5c2c8 ("fix: TreeNode has same attribute with NodeMixin"), which renamed `TreeNode.size` to `_size`. The traceback above is from the old `self.size = size` line. Checked master's `tools/size_report.py` with anytree 2.13.0 (current release) on an ESP32-S3 NuttX ELF: `size_report.py -k nuttx -z <dir> -o out all` writes rom.json and ram.json without errors. Can this be closed?

closed 2026-11-11 by xiaoxiang781216