apache/nuttx #11637: Listing all CONFIG_ symbols defined inside the source code but without corresponding "config SYMBOL" in Kconfig

acassis, 2024-02-01:
I decided to list all the symbols that are missing in our source code (thanks ChatGPT for help with the shell script added here).

Probably we will need a force task to look and fix these symbols, because they are more than 6K !!!!

[missing_symbols_in_kconfig.txt](https://github.com/apache/nuttx/files/14129608/missing_symbols_in_kconfig.txt)
[check_config_symbols.txt](https://github.com/apache/nuttx/files/14129616/check_config_symbols.txt)

acassis, 2024-02-01:
Note: you need to rename check_config_symbols.txt to check_config_symbols.sh and add permission execution to the file or run:
$ sh check_config_symbols.sh

anchao, 2024-02-18:
Some minor improvement:
1. Add Kconfig-* rules
2. Config name should also contain menuconfig ***
3. Enhance restriction rules to avoid adding incorrect configs

```
#!/bin/bash

# Set the root directory of your source tree
SOURCE_DIR="/home/archer/code/nuttx/n2/nuttx/"

# Find all Kconfig files in the source tree
KCONFIG_FILES=$(find "$SOURCE_DIR" -name "Kconfig*" -type f)

# Extract all CONFIG_ symbols generated from "config" statements in Kconfig files
CONFIG_SYMBOLS_KCONFIG=$(grep -o -h -E '^config [A-Za-z0-9_]*|^menuconfig [A-Za-z0-9_]*' $KCONFIG_FILES | awk '{print "CONFIG_"$2}' | uniq | sort -u)

# Extract all CONFIG_ symbols from source files
CONFIG_SYMBOLS_SOURCE=$(grep -o -h ' CONFIG_[A-Za-z0-9_]*' $(find "$SOURCE_DIR" -type f -name '*.[ch]') | sed 's/^ //g' | uniq | sort -u)

# Find CONFIG_ symbols that are in source code but not in Kconfig files
DIFF_SYMBOLS=$(comm -23 <(echo "$CONFIG_SYMBOLS_SOURCE") <(echo "$CONFIG_SYMBOLS_KCONFIG"))

# Output symbols that are in source code but not in Kconfig files
echo "CONFIG symbols in source code without corresponding 'config' statement in Kconfig:"
echo "$DIFF_SYMBOLS"
```