apache/nuttx #11643: ESP32-S3: chip/esp32s3_wlan.c:134:22: error: field 'rxb' has incomplete type

mirkokral, 2024-02-02:
With the wifi support, this error happens.
log:
```log
Create version.h
LN: platform/board to /home/mirko/nuttxspace/apps/platform/dummy
Register: coremark
Register: nsh
Register: sh
Espressif HAL for 3rd Party Platforms: initializing submodules...
Applying patches...
CC:  chip/esp32s3_wlan.c chip/esp32s3_wlan.c:134:22: error: field 'rxb' has incomplete type
  134 |   struct iob_queue_s rxb;
      |                      ^~~
chip/esp32s3_wlan.c:138:22: error: field 'txb' has incomplete type
  138 |   struct iob_queue_s txb;
      |                      ^~~
chip/esp32s3_wlan.c: In function 'wlan_cache_txpkt_tail':
chip/esp32s3_wlan.c:272:7: warning: implicit declaration of function 'iob_tryadd_queue' [-Wimplicit-function-declaration]
  272 |       iob_tryadd_queue(priv->dev.d_iob, &priv->txb);
      |       ^~~~~~~~~~~~~~~~
chip/esp32s3_wlan.c: In function 'wlan_recvframe':
chip/esp32s3_wlan.c:296:9: warning: implicit declaration of function 'iob_remove_queue' [-Wimplicit-function-declaration]
  296 |   iob = iob_remove_queue(&priv->rxb);
      |         ^~~~~~~~~~~~~~~~
chip/esp32s3_wlan.c: In function 'wlan_transmit':
chip/esp32s3_wlan.c:323:17: warning: implicit declaration of function 'iob_peek_queue' [-Wimplicit-function-declaration]
  323 |   while ((iob = iob_peek_queue(&priv->txb)) != NULL)
      |                 ^~~~~~~~~~~~~~
chip/esp32s3_wlan.c: In function 'wlan_ifup':
chip/esp32s3_wlan.c:818:3: warning: implicit declaration of function 'IOB_QINIT' [-Wimplicit-function-declaration]
  818 |   IOB_QINIT(&priv->rxb);
      |   ^~~~~~~~~
chip/esp32s3_wlan.c: In function 'wlan_ifdown':
chip/esp32s3_wlan.c:875:3: warning: implicit declaration of function 'iob_free_queue' [-Wimplicit-function-declaration]
  875 |   iob_free_queue(&priv->rxb);
      |   ^~~~~~~~~~~~~~
make[1]: *** [Makefile:144: esp32s3_wlan.o] Error 1
make: *** [tools/LibTargets.mk:164: arch/xtensa/src/libarch.a] Error 2
```

alandeassis, 2024-02-02:
Hi @mirkokral thank you for posting this issue. What board and board profile are you using? Seems like you forgot to enable something related to IOB.

@tmedicci did you know about this issue?

mirkokral, 2024-02-02:
Seems like if i make IOB_NCHAINS more than 0 then it works.
Also btw using a custom definition of the lilygo t-deck.

closed 2024-02-02 by mirkokral