apache/nuttx #11644: ESP32-S3: undefined reference to...

mirkokral, 2024-02-02:
While compiling for the ESP32-S3, i can't link.
[.config](https://pastebin.com/B1CvAHaS)
Log: ```log
mirko@mirkopc:~/nuttxspace/nuttx$ make -j4
Create version.h
Espressif HAL for 3rd Party Platforms: initializing submodules...
LN: platform/board to /home/mirko/nuttxspace/apps/platform/dummy
Applying patches...
Register: nsh
Register: sh
CC:  regex/regexec.c chip/esp32s3_wifi_adapter.c: In function 'esp_thread_semphr_get':
chip/esp32s3_wifi_adapter.c:1184:13: warning: implicit declaration of function 'task_tls_alloc' [-Wimplicit-function-declaration]
 1184 |       ret = task_tls_alloc(esp_thread_semphr_free);
      |             ^~~~~~~~~~~~~~
CC:  signal/sig_usleep.c chip/esp32s3_wifi_adapter.c:1194:17: warning: implicit declaration of function 'task_tls_get_value' [-Wimplicit-function-declaration]
 1194 |   sem = (void *)task_tls_get_value(wifi_task_key);
      |                 ^~~~~~~~~~~~~~~~~~
chip/esp32s3_wifi_adapter.c:1204:13: warning: implicit declaration of function 'task_tls_set_value' [-Wimplicit-function-declaration]
 1204 |       ret = task_tls_set_value(wifi_task_key, (uintptr_t)sem);
      |             ^~~~~~~~~~~~~~~~~~
CPP:  /home/mirko/nuttxspace/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld-> /home/mirko/nuttxspace/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/esp_rom/esp32s3/ld/esp32s3.rom.api.CPP:  /home/mirko/nuttxspace/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/soc/esp32s3/ld/esp32s3.peripherals.ld-> /home/mirko/nuttxspace/nuttx/arch/xtensa/src/chip/esp-hal-3rdparty/components/soc/esp32s3/ld/esp32s3.peripherals.LD: nuttx
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld: warning: /home/mirko/nuttxspace/nuttx/nuttx has a LOAD segment with RWX permissions
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld: esp32s3_start.o:(.iram1.literal+0x30): undefined reference to `xtensa_earlyserialinit'
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld: esp32s3_start.o: in function `__esp32s3_start':
/home/mirko/nuttxspace/nuttx/arch/xtensa/src/chip/esp32s3_start.c:348:(.iram1+0x9a): undefined reference to `xtensa_earlyserialinit'
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld: /home/mirko/nuttxspace/nuttx/staging/libarch.a(xtensa_initialize.o):(.literal.up_initialize+0x4): undefined reference to `xtensa_serialinit'
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld: /home/mirko/nuttxspace/nuttx/staging/libarch.a(xtensa_initialize.o):(.literal.up_initialize+0x8): undefined reference to `xtensa_netinitialize'
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld: /home/mirko/nuttxspace/nuttx/staging/libarch.a(xtensa_initialize.o): in function `up_initialize':
/home/mirko/nuttxspace/nuttx/arch/xtensa/src/common/xtensa_initialize.c:139:(.text.up_initialize+0x18): undefined reference to `xtensa_serialinit'
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld: /home/mirko/nuttxspace/nuttx/arch/xtensa/src/common/xtensa_initialize.c:144:(.text.up_initialize+0x1e): undefined reference to `xtensa_netinitialize'
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld: /home/mirko/nuttxspace/nuttx/staging/libdrivers.a(syslog_channel.o):(.literal.syslog_default_putc+0x0): undefined reference to `up_putc'
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld: /home/mirko/nuttxspace/nuttx/staging/libdrivers.a(syslog_channel.o): in function `syslog_default_putc':
/home/mirko/nuttxspace/nuttx/drivers/syslog/syslog_channel.c:229:(.text.syslog_default_putc+0x6): undefined reference to `up_putc'
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld: /home/mirko/nuttxspace/nuttx/staging/libarch.a(xtensa_nputs.o): in function `up_nputs':
/home/mirko/nuttxspace/nuttx/arch/xtensa/src/common/xtensa_nputs.c:44:(.text.up_nputs+0xa): undefined reference to `up_putc'
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld: /home/mirko/nuttxspace/nuttx/staging/libboard.a(esp32s3_board_tim.o):(.literal.board_tim_init+0x8): undefined reference to `esp32s3_timer_initialize'
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld: /home/mirko/nuttxspace/nuttx/staging/libboard.a(esp32s3_board_tim.o): in function `board_tim_init':
/home/mirko/nuttxspace/nuttx/boards/xtensa/esp32s3/common/src/esp32s3_board_tim.c:111:(.text.board_tim_init+0x8): undefined reference to `esp32s3_timer_initialize'
make[1]: *** [Makefile:174: nuttx] Error 1
make: *** [tools/Unix.mk:546: nuttx] Error 2
```

acassis, 2024-02-02:
I just tested here using the esp32s3-devkit with wifi board profile and it worked like a charm:

```
alan@dev:~/nuttxspace/nuttx$ minicom

Welcome to minicom 2.8

OPTIONS: I18n 
Port /dev/ttyUSB0, 12:23:12

Press CTRL-A Z for help on special keys

nsh> ?
help usage:  help [-v] [<cmd>]

    .           cd          exec        ifup        printf      time        
    [           cp          exit        kill        ps          true        
    ?           cmp         false       ls          pwd         truncate    
    alias       dirname     fdinfo      mkdir       rm          uname       
    unalias     dd          free        mkrd        rmdir       umount      
    arp         df          help        mount       set         unset       
    basename    dmesg       hexdump     mv          sleep       uptime      
    break       echo        ifconfig    nslookup    source      usleep      
    cat         env         ifdown      pidof       test        xd          

Builtin Apps:
    iperf    nsh      ping     rand     renew    sh       wapi                  
nsh> uname -a                                                                   
NuttX  12.4.0-RC0 e40b66bd6f Feb  2 2024 12:22:37 xtensa esp32s3-devkit         
nsh> ifconfig                                                                   
wlan0   Link encap:Ethernet HWaddr 7c:df:a1:e0:10:dc at UP mtu 1500             
        inet addr:10.0.0.2 DRaddr:10.0.0.1 Mask:255.255.255.0                   
                                                                                
nsh> wapi psk wlan0 mypassword 3                                                  
nsh> wapi essid wlan0 MyRouter 1                                                
nsh> renew wlan0                                                                
nsh> ifconfig                                                                   
wlan0   Link encap:Ethernet HWaddr 7c:df:a1:e0:10:dc at RUNNING mtu 1500        
        inet addr:192.168.0.19 DRaddr:192.168.0.1 Mask:255.255.255.0            
                                                                                
nsh> ping 8.8.8.8                                                               
PING 8.8.8.8 56 bytes of data                                                   
56 bytes from 8.8.8.8: icmp_seq=0 time=40.0 ms                                  
56 bytes from 8.8.8.8: icmp_seq=1 time=40.0 ms                                  
56 bytes from 8.8.8.8: icmp_seq=2 time=40.0 ms                                  
56 bytes from 8.8.8.8: icmp_seq=3 time=40.0 ms                                  
56 bytes from 8.8.8.8: icmp_seq=4 time=40.0 ms                                  
56 bytes from 8.8.8.8: icmp_seq=5 time=40.0 ms                                  
56 bytes from 8.8.8.8: icmp_seq=6 time=40.0 ms                                  
56 bytes from 8.8.8.8: icmp_seq=7 time=40.0 ms                                  
56 bytes from 8.8.8.8: icmp_seq=8 time=40.0 ms                                  
56 bytes from 8.8.8.8: icmp_seq=9 time=40.0 ms                                  
10 packets transmitted, 10 received, 0% packet loss, time 10100 ms              
rtt min/avg/max/mdev = 40.000/40.000/40.000/0.000 ms                            
nsh> 
```

Please see complete log and compilation attached!
[esp32s3_wifi_log.txt](https://github.com/apache/nuttx/files/14142097/esp32s3_wifi_log.txt)

mirkokral, 2024-02-02:
> I just tested here using the esp32s3-devkit with wifi board profile and it worked like a charm:
>
> ```
> alan@dev:~/nuttxspace/nuttx$ minicom
>
>
> Welcome to minicom 2.8
>
> OPTIONS: I18n
> Port /dev/ttyUSB0, [...]

What config do you use?

acassis, 2024-02-02:
WiFi config, as you can see inside esp32s3_wifi_log.txt file

acassis, 2024-02-02:
Ahh, maybe you are using the S3 before Espressif move the Native port to a port based on their HAL. Please clone a pristine nuttx and apps and use the esp32s3-devkit as reference.

mirkokral, 2024-02-02:
On the wifi config, i get only
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld:/home/mirko/nuttxspace/nuttx/boards/xtensa/esp32s3/esp32s3-devkit/../common/scripts/flat_memory.ld.tmp:1: undefined symbol `API_vhci_host_check_send_available' referenced in expression

mirkokral, 2024-02-02:
> Ahh, maybe you are using the S3 before Espressif move the Native port to a port based on their HAL. Please clone a pristine nuttx and apps and use the esp32s3-devkit as reference.

By the way, i am using the latest stable release.

mirkokral, 2024-02-02:
Works after config tweaks and updating nuttx.

closed 2024-02-02 by mirkokral