apache/nuttx #11645: esp32s3-devkit:usbnsh not compiling

mirkokral, 2024-02-02:
Log:
mirko@mirkopc:~/nuttxspace/nuttx$ ./tools/configure.sh esp32s3-devkit:usbnsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
CP: arch/dummy/Kconfig to /home/mirko/nuttxspace/nuttx/arch/dummy/dummy_kconfig
CP: boards/dummy/Kconfig to /home/mirko/nuttxspace/nuttx/boards/dummy/dummy_kconfig
LN: platform/board to /home/mirko/nuttxspace/apps/platform/dummy
LN: include/arch to arch/xtensa/include
LN: include/arch/board to /home/mirko/nuttxspace/nuttx/boards/xtensa/esp32s3/esp32s3-devkit/include
LN: drivers/platform to /home/mirko/nuttxspace/nuttx/drivers/dummy
LN: include/arch/chip to /home/mirko/nuttxspace/nuttx/arch/xtensa/include/esp32s3
LN: arch/xtensa/src/chip to /home/mirko/nuttxspace/nuttx/arch/xtensa/src/esp32s3
LN: arch/xtensa/src/board to /home/mirko/nuttxspace/nuttx/boards/xtensa/esp32s3/esp32s3-devkit/../common
LN: arch/xtensa/src/board/board to /home/mirko/nuttxspace/nuttx/boards/xtensa/esp32s3/esp32s3-devkit/src
#
# configuration written to .config
#
mirko@mirkopc:~/nuttxspace/nuttx$ make -j4
Create version.h
LN: platform/board to /home/mirko/nuttxspace/apps/platform/dummy
Register: nsh
Register: sh
LD: nuttx
/home/mirko/Documents/xtensa-esp-elf-13.2.0_20230928-x86_64-linux-gnu/xtensa-esp-elf/bin/xtensa-esp-elf-ld:/home/mirko/nuttxspace/nuttx/boards/xtensa/esp32s3/esp32s3-devkit/../common/scripts/flat_memory.ld.tmp:1: undefined symbol `API_vhci_host_check_send_available' referenced in expression
make[1]: *** [Makefile:174: nuttx] Error 1
make: *** [tools/Unix.mk:546: nuttx] Error 2

alandeassis, 2024-02-02:
ping @tmedicci @PilnyTomas

mirkokral, 2024-02-03:
I got it to compile with some (questionable) hacks that make incompatibility with every other board that is not xtensa.

mirkokral, 2024-02-03:
Works after pull.

closed 2024-02-03 by mirkokral